ngardiner/TWCManager #391: Problem with AddConsumptionOffset API Command

ChutneyMary, 2021-11-18:
Hi @ngardiner. I'm trying to integrate an unsupported home automation system (Loxone) with TWCManager. The home automation system supports sending POST commands, so I anticipate the best way will be using the AddConsumptionOffset command.

I've tried three independent methods to send the POST command -

1. Via the home automation in-built facility,
2. Via Node-Red,
3. Directly from the command line.

`curl -X POST -d '{ "offsetName": "WattsOffset", "chargeNowRate": -10000, "offsetUnit": "W" } http://10.1.1.21:8080/api/addConsumptionOffset`

I'm attempting to set the chargeNowRate to -10000. I anticipate I should see Generation reflect 10kW, but nothing changes.

Can you please confirm the facility does work?

There's a high likelihood I don't know how to properly format the POST command.

ngardiner, 2021-11-18:
Hi @ChutneyMary 

"chargeNowRate" should be "offsetValue" - it's nothing you did wrong, it's an error in the doco (looks like a copy / paste error) which I'll get fixed now, thanks for letting me know

ChutneyMary, 2021-11-18:
Thanks! Working fine now.

There's one more thing I hope you can help with... I'm trying to reduce the minimum charge rate to offer 5 amps (I'm on 3-phase on an Australian charger (and in Perth, where the grid operator keeps lines at about 255V), so still an acceptable charge rate.

I've read in a few areas that the minimum charge rate offered is 6 amps due to country settings. Is there somewhere I can change that value from 6 to 5? I've set minAmpsPerTWC to 5, but that doesn't seem to change anything?

ngardiner, 2021-11-18:
The European TWC (which is also the TWC used in AU) is limited to a minimum of 6A, there's no way we can convince the TWC itself to go lower than this as it's baked in firmware.

The recent introduction of API controlled charge rate means we will soon be able to set it to 5A on the vehicle side using the API. This isn't in place today.

ChutneyMary, 2021-11-19:
Thanks. For info, my old-ish 2015 Tesla Model S appears to have topped out with firmware at v2020.48.37.8. I don't think I'll see a firmware to support API controlled charge rate (but will be pleasantly surprised if it's released).

Very thankful for your TWCManager though!

closed 2021-11-19 by ChutneyMary

mikey4321, 2021-11-19:
According to https://www.teslafi.com/firmware.php about 40% of AP1 cars they are tracking have a 2021 firmware.  Suspect we just need to be patient or we all get it in this year's Christmas update. 

I'm on the same version as you - 2014 ap1 model S. 

[image]