ngardiner/TWCManager #395: Lost HeartBeat

Telectroboy, 2021-11-24:
Hello,

First, I'm new in TWCManager. Thanks a lot for this impressive job !!! :)

It's not really an issue it's a problem I meet.

I've installed TWCManager 1.22 then latest in a docker (with docker-composer) in a Debian. 
I use a ELFIN EW11A Wifi RS485 converter. The converter is set to 9600 bauds 8 1 N in netp connection.
[image]

[image]

I run the socat command : sudo socat pty,link=/dev/ttyUSB0,raw tcp:192.168.1.170:8899&
With minicom ou screen datas can be sent and received (with an other device connected to RS485)

TWCManager latest start with :
twcmanager_1  | 07:21:01 🎮 HTTP     20 Serving at port: 8080
twcmanager_1  | 07:21:01 ⛽ Manager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
twcmanager_1  | 07:21:01 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
twcmanager_1  | 07:21:01 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 07:21:01 ⛽ Manager  20 Set slave TWC 022C protocolVersion to 1, minAmpsTWCSupports to 5.
twcmanager_1  | 07:21:01 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 07:21:02 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 07:21:02 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72

This last info repeat 50 times in few seconds.

Then everything stops!

In the TWCManager web page I lose the heartbeat : 
[image]

I've been able after a big reset to start some charging sessions but they never last more than 3 minutes...
[image]

Do you have any suggestion?

Thanks

tjikkun, 2021-11-24:
Can you try without the `socat` and set `"port": "socket://192.168.1.170:8899"` in `config.json`?
Don't know if it will help, but maybe worth a try?

Telectroboy, 2021-11-24:
I need to change the config in docker-compose.yml file too.
 But I don't know what to put in place of so I've wrote # to comment those 2 lines.

It solved the heartbeat issue!!!

But I have something new : 

twcmanager_1  | 12:49:01 🎮 HTTP     20 Serving at port: 8080
twcmanager_1  | 12:49:01 ⛽ Manager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
twcmanager_1  | 12:49:01 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
twcmanager_1  | 12:49:02 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:02 ⛽ Manager  20 Set slave TWC 022C protocolVersion to 1, minAmpsTWCSupports to 5.
twcmanager_1  | 12:49:02 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:03 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:04 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:05 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:06 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:06 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:16 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:17 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:18 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:19 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:20 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:21 ⛽ Manager  20 32.00 amp slave TWC 022C is ready to link.  Sign: 72
twcmanager_1  | 12:49:25 ⛽ Policy   20 New policy selected; changing to Track Green Energy
twcmanager_1  | 12:49:25 ⛽ Slave    20 SHB 022C: 00 00.00/00.00A 0000  M: 05 00.00/00.00A 0000
twcmanager_1  | 12:49:25 ⛽ Slave    20 Slave power for TWCID 022C, status: 0

twcmanager_1  | 12:49:25 ⛽ Manager  20 BackgroundError: Traceback (most recent call last):
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 269, in background_tasks_thread
twcmanager_1  |     check_green_energy()
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 343, in check_green_energy
twcmanager_1  |     master.setConsumption(module["name"], module["ref"].getConsumption())
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/HASS.py", line 63, in getConsumption
twcmanager_1  |     self.update()
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/HASS.py", line 139, in update
twcmanager_1  |     self.consumedW = float(apivalue)
twcmanager_1  | TypeError: float() argument must be a string or a number, not 'NoneType'
twcmanager_1  | , occurred when processing background task

The background error is non stop repeating...
I imagine it's because I don't have any input value for the green power.

tjikkun, 2021-11-24:
You can try disabling the Energy -> HASS module in config.json
If you want the module enabled, you can set logLevel to 10, to see more output why it fails.

Telectroboy, 2021-11-24:
OK So it's normal.
Just because for the moment I don't have any energy info in Home Assistant !

Thanks a lot for your help.

:)))

Telectroboy, 2021-11-24:
Just for those who want to do the same. 
Here is the link to the converter ;
http://www.hi-flying.com/elfin-ew10-elfin-ew11

Very cheap 10€ on Aliexpress !

Telectroboy, 2021-11-24:
It doesn't work when the car is plugged... Same issue even if the heartbeat is good :(
And after 5 failures I need to unplug and plug back. To rearm it.

Telectroboy, 2021-11-24:
twcmanager_1  | 22:15:38 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:38 ⛽ Slave    20 SHB 022C: 02 16.77/00.00A 0000  M: 05 00.00/22.00A 0000
twcmanager_1  | 22:15:38 ⛽ Slave    20 Slave power for TWCID 022C, status: 2
twcmanager_1  | 22:15:38 ⛽ Manager  12 Rx@: (FE) FD E0 02 2C 77 77 02 00 00 06 96 00 00 9A
twcmanager_1  | 22:15:38 ⛽ Master   10 Total amps all slaves are using: 16.86
twcmanager_1  | 22:15:38 ⛽ Master   10 Number of cars charging now: 1
twcmanager_1  | 22:15:38 ⛽ Slave    10 desiredAmpsOffered TWC: 02 2C reduced from 22 to 22 with 1 cars charging and flex Amps of 0.0.
twcmanager_1  | 22:15:38 ⛽ Slave    10 TWCID=02 2C desiredAmpsOffered=22 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=22 self.reportedAmpsActual=16.86 now - self.timeReportedAmpsActualChangedSignificantly=0
twcmanager_1  | 22:15:38 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:39 RS485      12 Tx@: C0 FB E0 77 77 02 2C 05 08 98 00 00 00 00 00 00 A1 C0
twcmanager_1  | 22:15:40 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.1.144)
twcmanager_1  | 22:15:40 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.1.144)
twcmanager_1  | 22:15:40 ⛽ Master   10 Total amps all slaves are using: 16.86
twcmanager_1  | 22:15:40 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.1.144)
twcmanager_1  | 22:15:40 ⛽ Master   10 Number of cars charging now: 1
twcmanager_1  | 22:15:40 🎮 HTTP     10 Ending API GET (Url: /api/getStatus / IP: 192.168.1.144)
twcmanager_1  | 22:15:40 RS485      12 Tx@: C0 FB E0 77 77 02 2C 05 08 98 00 00 00 00 00 00 A1 C0
twcmanager_1  | 22:15:40 ⛽ Manager  12 Rx@: (FE) FD E0 02 2C 77 77 02 08 98 00 00 00 00 9E
twcmanager_1  | 22:15:40 ⛽ Master   10 Total amps all slaves are using: 0.0
twcmanager_1  | 22:15:40 ⛽ Master   20 Charge Session Stopped for Slave TWC 022C
twcmanager_1  | 22:15:40 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 22:15:40 ⛽ Slave    10 TWCID=02 2C desiredAmpsOffered=22 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=22 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=0
twcmanager_1  | 22:15:40 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:40 ⛽ Slave    20 SHB 022C: 02 00.00/22.00A 0000  M: 00 00.00/00.00A 0000
twcmanager_1  | 22:15:40 ⛽ Slave    20 Slave power for TWCID 022C, status: 2
twcmanager_1  | 22:15:40 ⛽ Manager  12 Rx@: (FE) FD E0 02 2C 77 77 02 08 98 00 00 00 00 9E
twcmanager_1  | 22:15:40 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 22:15:40 ⛽ Slave    10 TWCID=02 2C desiredAmpsOffered=22 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=22 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=0
twcmanager_1  | 22:15:40 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:41 RS485      12 Tx@: C0 FB E0 77 77 02 2C 00 00 00 00 00 00 00 00 00 FC C0
twcmanager_1  | 22:15:42 RS485      12 Tx@: C0 FB E0 77 77 02 2C 00 00 00 00 00 00 00 00 00 FC C0
twcmanager_1  | 22:15:42 ⛽ Manager  12 Rx@: (FE) FD E0 02 2C 77 77 02 08 98 00 00 00 00 9E
twcmanager_1  | 22:15:42 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 22:15:42 ⛽ Slave    10 TWCID=02 2C desiredAmpsOffered=22 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=22 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=2
twcmanager_1  | 22:15:42 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:42 ⛽ Manager  12 Rx@: (FE) FD E0 02 2C 77 77 02 08 98 00 00 00 00 9E
twcmanager_1  | 22:15:42 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 22:15:42 ⛽ Slave    10 TWCID=02 2C desiredAmpsOffered=22 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=22 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=2
twcmanager_1  | 22:15:42 ⛽ Slave    10 set_last_amps_offered(TWCID=02 2C, desiredAmpsOffered=22)
twcmanager_1  | 22:15:42 📊 HASS     13 Sending POST request to HomeAssistant for sensor sensor.twcmanager_022C_amps_in_use (value 0.0).

closed 2022-02-15 by ngardiner

Telectroboy, 2022-02-15:
Still Open... I haven't found any solution for this issue !

ngardiner, 2022-02-15:
Can you explain the issue in detail for me? Your car doesn't charge when plugged in?

Telectroboy, 2022-02-15:
Hello,
Yes, it's start to charge for 10 to 30 seconds and immediatly stop.
It seems the connection is lost or communication is not correct between TWCManager and the TWC.
 
Obviously, the charger and the car work in normal condition.

ngardiner, 2022-02-15:
That's possible, unfortunately the logs don't show anything out of the ordinary, they just show that the car started charging and then stopped charging. The most important clue would be the TWC itself, if there's a loss of connectivity the TWC will start flashing a red LED a number of times.

Could you give it a try and then let us know what the LED combination is? There's a table in the TWC manual which tells you what the LED combinations mean.

Telectroboy, 2022-02-15:
Yes you are right. It displayed a blinking red led.

ngardiner, 2022-02-15:
You need to note the pattern so you can identify what the issue is

[image]

If the issue is the 5 blink communication issue, you need to check the wiring / RS485 adapter. I say this because if you were losing connectivity, we would not just see charging vehicles go from 1 to 0, we'd see a loss of slave heartbeat and you'd see master/slave linkready messages in your logs. There's TX and RX evident in the logs which span only 4 seconds and during that time charging appears to stop, with no log entry suggesting we sent a stop command via the API, so it all seems external and we're just observing it.

That said, just because we can clearly see the TX/RX doesn't mean the other side can, it could be a loose connection for our TX to its RX (although we'd still see linkready messages if that was the case) or some other loss or corruption of the message.

From the logs you shared, we just see the TWC offering charge, and then no longer charging. No communications issues are evident.

Telectroboy, 2022-02-16:
Hello Nathan,

First of all, thank you for your help. I really appreciate :)

And sorry for the late answer. I didn't had time to check again. Everything is unplugged for the moment.
As far as I remember the failure displayed is 6 blinks so Overvoltage. 
Which doesn't make any sense...
I've tried with external power supply.
I've tried with better cable with and without terminaison resistor.
I've tried with wifi antenna far from the TWC.
But it doesn't make any difference. Still overvoltage failure. I need to try again maybe on thursday to understand.