ngardiner/TWCManager #399: no reaction to charge now, permission issue and understanding scheduling

flobbr, 2021-12-14:
Hi there,

i've had TWCManager run the whole summer without problem and tracking the solar production worked really really well.
Now i updated TWCManager to the latest version and also ran `make install`. There seems to be some issue while doing that because afterwards i get the follwing message in die webUI:

> We experienced an error trying to save your settings. This mea ns that while your current settings will be retained for the current session, th ey will not be saved if TWCManager exits. Please check the permissions on /etc/t wcmanager/settings.json and try saving your settings again.

Running `sudo chown -R pi:pi /etc/twcmanager` makes that message dissappear.
For the winter when there is basically never enough solar enegry in the system to charge the car, i want the TWC to always be ready to charge at 5A.
These are my settings now: [https://imgur.com/a/kkTVZya](url)
I don't know what i am doing wrong but the TWC doesn't want to charge.
Setting the "Charge now" option on the webUI doesn't seem to work either, as the "Amps offered" field always stays at 0.
What am i doing wrong?

Many thanks for your help and thank you so much for this wonderful script!

MikeBishop, 2021-12-20:
Having scheduled charging go midnight to midnight means it's zero-length, not 24 hours.  Instead of changing scheduled charging, I'd set your Non-Scheduled power action to be "Charge at specified Non-Scheduled Charge Rate." Thus, your TWC will offer 5A whenever it's not actively tracking solar.

However, that still will stop your cars from charging when there _might_ be solar but there's not enough power.  You might try increasing the `startStopDelay` in your config file to a very large number, which will basically never stop the car from charging even if TWCManager doesn't want to offer power.

ChutneyMary, 2021-12-21:
You might need to change 5A to 6A as minimum value. My model of TWC only works with 6A as a minimum. I believe this value changes due to country specific settings in the TWC.

flobbr, 2021-12-21:
Thanks for the answers!
As the screenshots show, "non scheduled power action" is set to "charge at apecific non scheduled charge rate" already. I will try the config option. 

Yes i can confirm that there are issues with the 5A setting as "charge now" only works from 6A upwards.

ngardiner, 2022-08-26:
Closing some stale issues, please feel free to re-open if still ongoing.