ngardiner/TWCManager #400: Always offering 6A

drexcia, 2022-01-09:
Hi All, 

This looks like a similar issue to #364 

I have just removed all of my docker images and installed the latest image as I have been having some issues generally (mainly trying to get the API so stop the car charging). I can see that the car is always being offered up 6A even though there is no solar surplus. 

[image]

[image]

Any ideas?

raffiniert, 2022-01-23:
same here :-(
this is pretty useless like this.

Did you find out something?

I can understand by the comments in config.json that a charge-level below x amps (depends on 1 or 3 phase-connection) is not efficient, but of course then the charge must be stopped...

drexcia, 2022-01-25:
Hi Raffiniert, I haven't managed to find anything. Experiencing a number of issues which is making the whole charge process unreliable so I am back to just normal scheduled charges for now

raffiniert, 2022-01-26:
yeah. So much time wasted for my special installation hardware-wise and now just basically not using it AT ALL.

drexcia, 2022-02-23:
Closing this one, no longer occurring after I am manually entering in the refresh/access tokens manually.

closed 2022-02-23 by drexcia

raffiniert, 2022-02-24:
not working for me, please specify what exactly you changed. Of course I also entered the tokens!
Still it is always charging with at least 6A, even at midnight when set to solar surplus.

drexcia, 2022-02-24:
Are you generating new tokens every 8 hours and entering them into the settings page? I thought they lasted longer (confusion on my part). I’ll enter in new tokens in the morning around 8am which take care of the excess Solar that day.

raffiniert, 2022-02-24:
I entered new tokens and it did not shut down charing 1 minute later.
And on top, this is a joke, having to reenter the tokens all the time. I thought that's what the refresh-tokens were supposed to do?

drexcia, 2022-02-24:
It's not a joke, you just need to appreciate that this is free software
that people spend a lot of their own time on. There's a well documented
issue relating to tokens refreshing with the Tesla API that is being worked
on. I believe a fix is due in the coming weeks.

I plugged my car in tonight and it started charging at 6A with
nothing being offered. I've gone to the settings page and entered in new
tokens that are valid for 8 hours using Auth for Tesla. Those tokens kicked
in and my car stopped charging after about 5 minutes because I am set to
charge off solar surplus only. It needs longer than a minute to get it to
stop. Give it at least 5-10.

ngardiner, 2022-02-24:
@drexcia I will have something for you **very** soon, and thank you for the kind words

ngardiner, 2022-02-24:
@drexcia commit 398962b on the main / development branch just now is a complete replacement of the Tesla API login logic to allow login from web browser once again using the URL trick that a lot of sites now use. This will cut down the effort in updating Tesla API credentials considerably. It's step 1 of 2, now that this is in place I'll be doing some thorough testing of the refresh token flow to see if I can identify any issues

drexcia, 2022-02-24:
Fantastic! Will grab it now. Thank you :)