ngardiner/TWCManager #403: No module named TWCManager.__main__

raffiniert, 2022-01-17:
Hey

I just installed everything according to https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md (because docker-installation did not work, either).

When trying to start with `sudo -u twcmanager python -m TWCManager`, I get:

`/usr/bin/python: No module named TWCManager.__main__; 'TWCManager' is a package and cannot be directly executed`

At first, I also got a warning because there was no user named "twcmanager" but I created one. Do I need to ssh with that user as well? 

Sorry, this is actually my first day on a Raspberry :-)

Cheers
Raphi

EdwinGH, 2022-01-17:
You have to be in the TWCManager / lib directory to start it manually. Then it will look for the Package in the TWCManager / lib / TWCManager directory.

MiRoCZ, 2022-01-30:
I have same issues as reported @raffiniert  with Docker installation and now I get stucked on this issue in Manual instalation. Strictly followed instructions, but get same error as reported above. Did you please resolve issue and make TWCManager work ? Any hint or advice very appreciated.

ngardiner, 2022-02-05:
User ```twcmanager``` not existing is an issue as this is added as part of the ```sudo make install``` step of the installation.

Are you getting an error running this command? It doesn't make sense that the user account does not exist unless there is an error thrown during the make command as I have confirmed right now it is part of the makefile and has been for a long time now

ngardiner, 2022-02-06:
The setuptools team have deprecated setup.py and changed the way that python dependencies are installed

As a result I have had to make significant changes which now break backwards compatibility with Python < 3.6. I'm closing this issue off as it will no longer be relevant after these changes.

closed 2022-02-06 by ngardiner