ngardiner/TWCManager #404: Docker Image:Latest - commentjson missing?

VIDGuide, 2022-01-18:
Did an update to my stack today, pulled latest image, and now the app won't start, and it's logging over and over:
Traceback (most recent call last):
  File "TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 30, in <module>
    import commentjson
ModuleNotFoundError: No module named 'commentjson'

looks like this module is missing from the docker image?

ngardiner, 2022-01-22:
Thanks for letting me know! Unfortunately the entire build pipeline has been down for the last week due to an issue but I have spent time fixing it today and have it hopefully up and running again, so I'm doing a rebuild of the latest image now to see what's going on and will let you know what I find.

ngardiner, 2022-01-22:
I've rebuilt :latest and confirmed that ```commentjson``` was built as part of the image - could you please try pulling it again and see if it works for you now?

VIDGuide, 2022-01-23:
I'm afraid not. I did a normal pull/update cycle, and no change, so I did a full clear of cached docker images, and ensured it downloaded a fresh version, same result:
```
Traceback (most recent call last):
  File "TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 30, in <module>
    import commentjson
ModuleNotFoundError: No module named 'commentjson'
```

Weird thing is, I can even see the build script on dockerhub that shows the commentjson line being installed as part of the build, so I can't imagine why it's not there! (or why no-one else is reporting the issue)

But if I put tag #v1.2.3 in the compose file and run again, it will start up just fine.

ngardiner, 2022-02-15:
Hi @VIDGuide - I know it has been a long time since you reported this issue and my apologies, however this should now be fixed in v1.2.4

closed 2022-02-25 by ngardiner