ngardiner/TWCManager #406: Installation issues (software)

raffiniert, 2022-01-19:
Hey

I try to install the docker-version on a brand new Raspberry Pi Zero W.
I flashed the SD-card with the official tool with Raspbian OS Lite (both legacy "Buster" and newest "Bullseye") and execute each command as written in:

https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Docker.md

using the "stable" version.

- [x] sudo apt-get update
- [x] sudo apt-get install -y docker.io
- [ ] pip3 install docker-compose
first problem here: `-bash: pip3: command not found`

So I do (probably needed because I'm on the "lite" version?)

- [x] sudo apt install python3-pip

and then again 
- [x] pip3 install docker-compose
after that, "Kernel is not expected version, need to reboot", so:
- [x] sudo reboot

ok, let's try again:
- [x] pip3 install docker-compose
I get those warnings, despite running completely:
`The script distro is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script dotenv is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script jsonschema is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script docker-compose is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.`

I first ignored this and did:

- [ ] curl https://raw.githubusercontent.com/ngardiner/TWCManager/main/contrib/docker/docker-compose-v1.2.2.yml -o docker-compose.yml
but I got `sudo: docker-compose: command not found` which corresponds to the above warning, I guess

So I do
- [x] export PATH=/home/pi/.local/bin:$PATH 
- [x] sudo reboot
- [ ] curl https://raw.githubusercontent.com/ngardiner/TWCManager/main/contrib/docker/docker-compose-v1.2.2.yml -o docker-compose.yml
but I got `sudo: docker-compose: command not found` 

I don't know what to do anymore...

raffiniert, 2022-01-19:
oh, and running `docker run hello-world` as a test gives:

`docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.`

raffiniert, 2022-01-19:
ok, the documentation is probably wrong...

I had to run pip3 install with sudo as well

raffiniert, 2022-01-19:
ok, I can (kind of) start the docker container now, but no log output visible.

`docker container ls` is giving me:
CONTAINER ID        IMAGE                          COMMAND                  CREATED             STATUS                           PORTS               NAMES
2d70fa9cd4e4        twcmanager/twcmanager:v1.2.2   "./entrypoint.sh /us…"   4 minutes ago       Restarting (139) 7 seconds ago                       pi_twcmanager_1

while "restarting" is always reset to 0 after a short time, so it seems something is unhappy...

wooter, 2022-01-20:
Maybe try to install docker and docker-compose through following this manual:

https://devdojo.com/bobbyiliev/how-to-install-docker-and-docker-compose-on-raspberry-pi

raffiniert, 2022-01-20:
but, docker is installed now... please have a look at my last comment... :-) thanks!

wooter, 2022-01-20:
Obviously something is not going right.

`sudo docker-compose -f docker-compose.yml up` should be happy purring along if everything is right.

raffiniert, 2022-01-20:
with `docker ps -a` I get:

`CONTAINER ID        IMAGE                          COMMAND                  CREATED             STATUS                            PORTS               NAMES
db952d32749d        twcmanager/twcmanager:v1.2.2   "./entrypoint.sh /us…"   2 minutes ago       Restarting (139) 45 seconds ago                       pi_twcmanager_1`

it's basically restarting all the time.

It is NOT connected to a TWC yet, as this is outside, it's freezing and wet and I want to get this working first - don't think this is the problem?

raffiniert, 2022-01-20:
really? What is the "dummy" interface for then?

wooter, 2022-01-20:
Which dummy interface?

docker-compose.yml specifies `/dev/ttyUSB0:/dev/ttyUSB0` and thats not dummy.

raffiniert, 2022-01-20:
I thought about this one: https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Interface_Dummy.md but I don't know how to enable it.

Btw, my docker-compose.yml specifies:

'devices:
      - "/dev/serial0:/dev/serial0"'

because I am using an RS485 CAN HAT, but I'm not sure if this is correct.

wooter, 2022-01-20:
As long as you can't communicate to your TWC, it won't _work_. It needs to communicate with your TWC to work, otherwise the service will just keep restarting.

raffiniert, 2022-01-20:
ok, thanks so much for your help. I will see if I can somehow safely attach it to the TWC, it's a pretty special installation :-)
Will solder the uFL-connector to the Raspi now, no chance to connect otherwise over that distance.

Do you happen to know how to change a file inside the docker container? I need to change Fronius.py to accomodate my 2 Fronius inverters: https://github.com/ngardiner/TWCManager/issues/380#issuecomment-961820043

wooter, 2022-01-20:
I think if you want to do that, you're better off with a manual installation. Docker is specifically aimed at running it out of the box.

https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md

raffiniert, 2022-01-20:
or I could fork it to my own github-repo, change that file and pull from there?

wooter, 2022-01-20:
Sure, could do too. But I'd suggest the manual installation for that too then.

raffiniert, 2022-01-20:
thanks a lot!!

closed 2022-02-06 by ngardiner