ngardiner/TWCManager #407: Product Announcement: Distributed Charge [beta] Board A0

AndySchroder, 2022-01-24:
Hello,

Not sure if there is a mailing list for TWCManager, so I'll just create this announcement as an issue....

I'd like to announce Distributed Charge [beta] Board A0:

http://andyschroder.com/DistributedCharge/BoardA0/Overview/ .

Board A0 coupled with a TOFU raspberry pi compute module 4 carrier board provides a super clean installation with a Gen2 Wall Connector. A single CAT5 wire is required to connect to the Gen 2 Wall Connector which provides power, RS-485 connection, AND a magical CAN bus connection to the charge port of the car.

The assembly can also be installed inside the car's trunk to allow bitcoin lightning network payments for energy that is delivered through the Gen2 Wall Connector. The second CAN bus on board A0 connects to high voltage battery to also monitor the charge current and voltage to verify that the energy that is being paid for to the wall connector over the bitcoin lightning network is actually being delivered.

Board A0 also have some extra relays, analog inputs, and GPIO conveniently available on rugged screw connectors for doing other fun stuff you might have in mind. There are also dedicated pins for UART so that you can do debugging over a USB serial adapter if you don't have an ethernet connection.

VIDGuide, 2022-01-24:
I must say this is a fascinating project on a larger scale!

I don't specifically have a need for the Bitcoin payment side of things, but I am very interested to the can bus access over the charge cable and what we can get/report on via that. (I love data!)

I see no wiring/assembly diagrams yet on the project, but I'll be keeping an eye out, that part is very much of interest to me!

AndySchroder, 2022-01-24:
I have recorded an installation time lapse video for the wall connector but have not yet had time to edit and publish it. Will report back once that is complete.

The bitcoin lightning network is going to be necessary to have a truly open payment method to allow for bi-directional and peer to peer net metering everywhere.

AndySchroder, 2022-08-24:
I decided not to publish the time lapse video, but instead have created the following web page which describes my setup with the Gen2 Wall Connector in detail: http://andyschroder.com/DistributedCharge/EV/TeslaGen2WallConnector/ .

MikeBishop, 2026-10-26:
Thanks for sharing. Product announcements are better suited to Discussions than issues, so I'm closing this here.

closed 2026-10-26 by MikeBishop