ngardiner/TWCManager #409: Installation instructions not correct/updated - not possible to install TWC manager

MiRoCZ, 2022-02-04:
Dear @ngardiner, with all respect for all work already done on this project, I´d like to ask you to update installation instructions with DETAILED step by step instructions, how to succesfully install TWC manager or please provide SD card image for RPi with functional TWCManager for download.  THANK YOU VERY MUCH !

ngardiner, 2022-02-05:
> Dear @ngardiner, with all respect for all work already done on this project, I´d like to ask you to update installation instructions with DETAILED step by step instructions, how to succesfully [...]

Hi @MiRoCZ,

What part of the instructions are you having difficulty with?

MiRoCZ, 2022-02-05:
Hi Nathan @ngardiner,
trying to strictly follow instalation instructions provided at https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md

but always ended with some kind of error at the final stage.

For manual installation of stable version after executing command "sudo make install" I just ended with error:

ModuleNotFoundError: No module named 'setuptools_rust'
make: *** [Makefile:66: setup] Error 1

Another manual installation of main development branch ended with following error:

pi@TWCManager:~/TWCManager $ sudo -u twcmanager python -m TWCManager
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
ImportError: No module named TWCManager.TWCManagerls

Docker installation  as per instruction was also unsuccessfull. 

I apologize if I ignore something common what is not mentioned in the instructions.

In order to avoid any issues I have recommended hardware :

RPI Zero WH, new 32 GB microSD card Verbatim Premium, USB to RS 485 CH 340C, Tesla Wall Connector Gen. 2 in Slave mode (F position), 3 phase 240V connection, RS-485 connection polarity correct,

Latest Raspberry Pi Imager v1.6.2  used to create image on new 32GB microSD 

OS selected: Raspberry PI OS (Legacy) - Released 2022-01-28
 (also tested with full RASPEBBRY PI OS (32bit)

SSH and VNC enabled

Hostname: TWCManager
User: pi (also tried twcmanager) - 
is it OK to use user pi for installation  or user twcmanager must be created and used for installation?  

Is my setting wrong or what is preventing to sucessfully install TWCManager ?

Thank you very much.
MiRo

ngardiner, 2022-02-05:
Hi MiRo,

It looks like your system python version is 2.7 which is an issue as the minimum we support is 3.4.

Could you please share the output of:

```
sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 2

python -V
```

MiRoCZ, 2022-02-05:
Hi Nathan,

I´m aware of min. 3.4 version requirement and my installation has Python 3.7.3
[image]

pi@TWCManager:~/TWCManager $ python -v
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'zipimport' # <class '_frozen_importlib.BuiltinImporter'>
# installed zipimport hook
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import _thread # previously loaded ('_thread')
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import _weakref # previously loaded ('_weakref')
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.7/encodings/__pycache__/__init__.cpython-37.pyc matches /usr/lib/python3                                                          .7/encodings/__init__.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/__init__.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/codecs.cpython-37.pyc matches /usr/lib/python3.7/codecs.py
# code object from '/usr/lib/python3.7/__pycache__/codecs.cpython-37.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0xb6701630>
# /usr/lib/python3.7/encodings/__pycache__/aliases.cpython-37.pyc matches /usr/lib/python3.                                                          7/encodings/aliases.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/aliases.cpython-37.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0xb6713                                                          190>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0xb6701310>
# /usr/lib/python3.7/encodings/__pycache__/utf_8.cpython-37.pyc matches /usr/lib/python3.7/                                                          encodings/utf_8.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/utf_8.cpython-37.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0xb6717b1                                                          0>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.7/encodings/__pycache__/latin_1.cpython-37.pyc matches /usr/lib/python3.                                                          7/encodings/latin_1.py
# code object from '/usr/lib/python3.7/encodings/__pycache__/latin_1.cpython-37.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourceFileLoader object at 0xb671b                                                          130>
# /usr/lib/python3.7/__pycache__/io.cpython-37.pyc matches /usr/lib/python3.7/io.py
# code object from '/usr/lib/python3.7/__pycache__/io.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/abc.cpython-37.pyc matches /usr/lib/python3.7/abc.py
# code object from '/usr/lib/python3.7/__pycache__/abc.cpython-37.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0xb671b3b0>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0xb671b210>
# /usr/lib/python3.7/__pycache__/site.cpython-37.pyc matches /usr/lib/python3.7/site.py
# code object from '/usr/lib/python3.7/__pycache__/site.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/os.cpython-37.pyc matches /usr/lib/python3.7/os.py
# code object from '/usr/lib/python3.7/__pycache__/os.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/stat.cpython-37.pyc matches /usr/lib/python3.7/stat.py
# code object from '/usr/lib/python3.7/__pycache__/stat.cpython-37.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0xb672de30>
# /usr/lib/python3.7/__pycache__/posixpath.cpython-37.pyc matches /usr/lib/python3.7/posixp                                                          ath.py
# code object from '/usr/lib/python3.7/__pycache__/posixpath.cpython-37.pyc'
# /usr/lib/python3.7/__pycache__/genericpath.cpython-37.pyc matches /usr/lib/python3.7/gene                                                          ricpath.py
# code object from '/usr/lib/python3.7/__pycache__/genericpath.cpython-37.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0xb66b5dd0>
import 'posixpath' # <_frozen_importlib_external.SourceFileLoader object at 0xb672df30>
# /usr/lib/python3.7/__pycache__/_collections_abc.cpython-37.pyc matches /usr/lib/python3.7                                                          /_collections_abc.py
# code object from '/usr/lib/python3.7/__pycache__/_collections_abc.cpython-37.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0xb66bc2                                                          30>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0xb67277d0>
# /usr/lib/python3.7/__pycache__/_sitebuiltins.cpython-37.pyc matches /usr/lib/python3.7/_s                                                          itebuiltins.py
# code object from '/usr/lib/python3.7/__pycache__/_sitebuiltins.cpython-37.pyc'
[... cut ...]
# extension module 'readline' executed from '/usr/lib/python3.7/lib-dynload/readline.cpytho                                                          n-37m-arm-linux-gnueabihf.so'
import 'readline' # <_frozen_importlib_external.ExtensionFileLoader object at 0xb66e3270>
import 'atexit' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.7/__pycache__/rlcompleter.cpython-37.pyc matches /usr/lib/python3.7/rlco                                                          mpleter.py
# code object from '/usr/lib/python3.7/__pycache__/rlcompleter.cpython-37.pyc'
import 'rlcompleter' # <_frozen_importlib_external.SourceFileLoader object at 0xb66e3350>
>>>

ngardiner, 2022-02-05:
Look at the output above when you ran the command manually. It is using python 2.7.

MiRoCZ, 2022-02-05:
To be honest, I don´t know why python 2.7 is used when result for command "python -v" is Python 3.7.3 as you can see on screenshot above.
But this is a result when I strictly followed installation instructions and used latest version of Raspberry PI OS Lite (legacy). Is this Raspbian version OK for TWCManager?

What exact version of Raspbian do you recommend for TWCManager ? Please confirm if user "pi" is recommended to be used for instalation? I´d like to make identical installation as you have in order to make TWCManager up and running.

MiRoCZ, 2022-02-05:
[image]

I´m using default user "pi" and user "twcmanager" is not created.

ngardiner, 2022-02-05:
User ```twcmanager``` is created by running ```make install``` per the documentation.

MiRoCZ, 2022-02-05:
As a "sudo make install" not finished completelly and ended with error, user twcmanager was not probably created. 
I don´t want to waste your time for this issue troubleshooting - why system with Python 3.7.3. installed is using 2.7 instead ?!?
 
Please let me know type of hardware and exact version of Raspbian you are using and system setting before you start clean installation of TWCManager as per instructions and I´ll try another clean installation tomorrow. Thank you and have a nice weekend.

ngardiner, 2022-02-05:
Closing this issue unless you are going to provide the error you get with a make install.

closed 2022-02-05 by ngardiner