ngardiner/TWCManager #413: Need to start TWCManager twice in order to make it work

MiRoCZ, 2022-02-20:
Hello, does anybody else also experience this behaviour when starting TWCManager ?

As you can see on the screenshot, when I execute 
sudo -u twcmanager python -m TWCManager 
for first time it ends in the loop of message below until interupted and TWC Manager doesn´t work

11:35:41 ⛽ Manager  20 32.00 amp slave TWC 2510 is ready to link.  Sign: A8

When interupted and immediatelly executed again
sudo -u twcmanager python -m TWCManager 
TCM Manager start working - Track Green Policy selected and everything is fine.

When I try to run TWCManager as process on the background (as per instructions) it behave same way and only message loop 
 ⛽ Manager  20 32.00 amp slave TWC 2510 is ready to link.  Sign: A8
visible when executed 
journalctl -f
and TWC Manager doesn´t work.

[image]

ngardiner, 2022-02-22:
No, I haven't seen this before, it means that your RX is okay but for some reason your TX is not.

Can you see your RS485 adaptor when this happens? I'd be interested in if you see the TX line stuck on or if it's blinking throughout. Also just to confirm you definitely have an RS485 and not RS232 adaptor? RS485 is half-duplex and doesn't require RTS/CTS signaling whereas RS232 is full duplex and does.

ngardiner, 2022-02-22:
Also I note that even when it does work your log has a message about an RX of unexpected length. You seem to have some signaling issues there, it's probably timing more than anything else that is making it work, how long is the run between your RS485 adaptor and your TWC?