ngardiner/TWCManager #415: HASS EMS plugin doesn't seem to work in v1.2.4

harmenwessels, 2022-02-22:
I started using TWCManager as of today and I can't seem to get the HASS EMS plugin to do anything.

I have the RS485 communication working with one TWC and also sending the sensor values to HASS is working. I also see these actions in the logging of the docker container.

I set the logLevel to 13 (i.s.o. 20) to see what the container is doing, I see a lot happing, but nothing from the HASS EMS plugin.

Is this a known issue? Or am I missing something?

Configuration section for HASS EMS (where of course I set the api key to a very long string which I won't paste here :)):

        "HASS": {
      # The HASS module allows fetching of consumption and generation statistics from HomeAssistant sensors.
            "enabled": true,

      # The IP address and port of the HomeAssistant front-end
            "serverIP": "192.168.1.25",
            "serverPort": "8123",
            "useHttps": false,

      # To obtain a HASS API key, via browser, click on your user profile, and
      # add a Long-Lived Access Token. Place it in the following variable:
            "apiKey": "<my-api-key>",

      # For HomeAssistant, the two settings below must be customized to point to the specific sensor
      # names you use within HomeAssistant. There is no default or common value for this, so it will
      # require customization to work correctly.
      # If you do not track one of these values (generation or consumption) via HASS, leave the parameter
      # blank, and it will not be retrieved.
            "hassEntityConsumption": "sensor.power_consumption",
            "hassEntityGeneration":  "sensor.power_production"
        }

harmenwessels, 2022-02-22:
I changed te logLevel to 10 to see if I get more output, but I don't.

When I grep the log for HASS i get the following, which means the module is registered, but only the HASSStatus module is actually doing something I am correct.
[image]

harmenwessels, 2022-02-24:
Some more input, I removed the uncommented the server and port from the config to see whether it would fail (just to make sure that part of the config is actuallty parsed). And it does fail, also in the EMS/HASS code. So it looks like the code is executed but nothing is logged?:

[image]

harmenwessels, 2022-02-24:
This is really strange, I just changed back the config, restarted the docker container and now I do see output in the logging:

[image]

I do notice it thinks it received watts i.s.o. the kW which are the default in Home Assistant since the introduction of the energy dashboard.

harmenwessels, 2022-02-24:
Is it true that the EMS system is only read if the "Non-Scheduled power action:" is set to "Track Green Energy"? Cause it only seems to work if I set this setting to "Track Green Energy".

MikeBishop, 2022-02-25:
More specifically, EMS is only polled when the policy has a `"background_task": "checkGreenEnergy",` option, which is only true for Track Green Energy and the corresponding Non Scheduled Power option.

If you want EMS polled, you could add a custom policy with this background task; it will only set your charging speed if the policy does not specify a `charge_amps` value.

closed 2022-02-26 by harmenwessels