ngardiner/TWCManager #417: PowerWall2 Module causes a lot exceptions if unable to connect (e.g. wrong IP)

GMerg, 2022-02-24:
Hello,

recently I logged in to my pi to update the TWCManager and to my surprise I saw it is throwing out a lot of errors.
I found out that the core problem was in the TeslaPowerwall2.py module where it was unable connect to my PowerWall2 due to an IP change.

However, there was no clear error shown in my log but a lot of exceptions caused by variables in my custom policies not being resolved.

It would be nice if you could implant some check about this.

`20:51:41 ⛽ Manager  40 ImportError: No module named 'psycopg2', when importing Vehicle.TeslaMateVehicle, not using Vehicle.TeslaMateVehicle
20:51:42 ⛽ Manager  40 ImportError: No module named 'sysv_ipc', when importing Control.WebIPCControl, not using Control.WebIPCControl
20:51:42 🎮 HTTP     20 Serving at port: 8080
20:51:42 ⛽ Manager  40 ImportError: No module named 'pyModbusTCP', when importing EMS.Kostal, not using EMS.Kostal
20:51:42 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
20:51:43 ⛽ Manager  20 32.00 amp slave TWC 0441 is ready to link.  Sign: 2D
20:51:43 ⛽ Manager  20 Set slave TWC 0441 protocolVersion to 2, minAmpsTWCSupports to 6.
Traceback (most recent call last):
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 1098, in <module>
    slaveTWC.receive_slave_heartbeat(heartbeatData)
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 657, in receive_slave_heartbeat
    self.master.getModuleByName("Policy").setChargingPerPolicy()
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 161, in setChargingPerPolicy
    policy["match"], policy["condition"], policy["value"]
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 347, in checkConditions
    if self.doesConditionMatch(match, condition, value, exitOn) == exitOn:
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 320, in doesConditionMatch
    return True if matchValue >= value else False
TypeError: '>=' not supported between instances of 'str' and 'int'
20:51:50 ⛽ Manager  20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 1098, in <module>
    slaveTWC.receive_slave_heartbeat(heartbeatData)
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 657, in receive_slave_heartbeat
    self.master.getModuleByName("Policy").setChargingPerPolicy()
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 161, in setChargingPerPolicy
    policy["match"], policy["condition"], policy["value"]
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 347, in checkConditions
    if self.doesConditionMatch(match, condition, value, exitOn) == exitOn:
  File "/home/pi/TWCManager/lib/TWCManager/Policy/Policy.py", line 320, in doesConditionMatch
    return True if matchValue >= value else False
TypeError: '>=' not supported between instances of 'str' and 'int'

20:51:55 ⛽ Manager  20 32.00 amp slave TWC 0441 is ready to link.  Sign: 2D
Traceback (most recent call last):
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 1098, in <module>
    slaveTWC.receive_slave_heartbeat(heartbeatData)
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 662, in receive_slave_heartbeat
    flex = self.master.getAllowedFlex()
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 272, in getAllowedFlex
    return self.allowedFlex
AttributeError: 'TWCMaster' object has no attribute 'allowedFlex'
20:51:56 ⛽ Manager  20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 1098, in <module>
    slaveTWC.receive_slave_heartbeat(heartbeatData)
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 662, in receive_slave_heartbeat
    flex = self.master.getAllowedFlex()
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 272, in getAllowedFlex
    return self.allowedFlex
AttributeError: 'TWCMaster' object has no attribute 'allowedFlex'
`

MikeBishop, 2022-02-25:
Hm, that looks like it might actually be a couple of related bugs; those properties shouldn't return a string under any circumstances.  I'll try to take a look at it soon.

GMerg, 2022-02-27:
OK, let me know if I can help. Enabling this module with a bogus IP should however be enough to replicate this.

MikeBishop, 2026-10-26:
The Powerwall2 module now returns the last cached values or numeric defaults when the gateway can't be reached, rather than raising. Could you re-verify on v1.4.0 (for example with the gateway unreachable) whether the exceptions are gone?

MikeBishop, 2026-10-30:
The Powerwall module now keeps the last reading (or numeric defaults) and logs the failure instead of raising when the Powerwall or gateway can't be reached, so a wrong or unreachable IP no longer floods the log with exceptions. This ships in 1.4.1; please reopen if you still see it.

closed 2026-10-30 by MikeBishop