ngardiner/TWCManager #420: MQTT Sensor support for reading generation or consumption data

KirkKirk, 2022-03-02:
@ngardiner, any chance to be implemented, please? I think it will be very useful.

       "MQTT": {
      # MQTT Sensor support for reading generation or consumption data.
      # This is not yet available.
          "enabled": false
        },

KirkKirk, 2022-03-06:
@ngardiner thanks for adding the module.
I pull it and if you need it, am happy to help with testing.

KirkKirk, 2022-07-27:
@ngardiner I noticed that you improve some other things and thanks for it. After an update and a pull, I got new errors on the Mqqt read (when "enabled": true,).

pi@TWCManager:~/TWCManager $ sudo -u twcmanager python -m TWCManager
00:19:20 🎮 HTTP     20 Serving at port: 8080
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 616, in <module>
    modinstance = modclassref(master)
  File "/home/pi/TWCManager/lib/TWCManager/EMS/MQTT.py", line 62, in __init__
    self.brokerIP, port=self.brokerPort, keepalive=30
AttributeError: 'MQTT' object has no attribute 'brokerPort'
pi@TWCManager:~/TWCManager $

mauriceflier, 2022-08-04:
I tried to configure this using the following entry in my config:

```
        "MQTT": {
          "enabled": true,
          "brokerIP": "192.168.x.x",
          "brokerPort": "1883",
          "serverPort": "1883",
      # The MQTT EMS module subscribes to the below topics to read Consumption/Generation values
      #    "username": "mqttuser_twc",
      #    "password": "mqttpass_twc",
          "topicConsumption": "homeassistant/sensor/twc/electricity_sum",
          "topicGeneration": "homeassistant/sensor/twc/grid_power"
        },
```
But i get the following error:
```
  File "/usr/src/TWCManager/lib/TWCManager/EMS/MQTT.py", line 62, in __init__
    self.brokerIP, port=self.brokerPort, keepalive=30
AttributeError: 'MQTT' object has no attribute 'brokerPort'
```

in **TWCManager/EMS/MQTT.py** ems-module mqtt-class "serverPort" is defined (as 8123, which is HomeAssistant"), but no "brokerPort" is defined.

from line 52: 
```
        logger.debug("Attempting to Connect to MQTT Broker")
        if self.brokerIP:
            self.__client = self.mqtt.Client("MQTT.EMS")
            if self.username and self.password:
                self.__client.username_pw_set(self.username, self.password)
            self.__client.on_connect = self.mqttConnect
            self.__client.on_message = self.mqttMessage
            self.__client.on_subscribe = self.mqttSubscribe
            try:
                self.__client.connect_async(
                    self.brokerIP, port=self.brokerPort, keepalive=30
```
could it be that brokerport needs to be defined in the MQTT Class?

reklame33, 2022-12-28:
Did you make TWCM MQTT read generated and consumed?

RichieB2B, 2023-05-14:
> But i get the following error:
> 
> ```
>   File "/usr/src/TWCManager/lib/TWCManager/EMS/MQTT.py", line 62, in __init__
>     self.brokerIP, port=self.brokerPort, keepalive=30
> AttributeError: 'MQTT' object has no attribute 'brokerPort'
> ```

I ran into the same issue and fixed this in #508

KirkKirk, 2023-05-23:
@RichieB2B it is working !!!
Thanks