ngardiner/TWCManager #421: Power vs. Phases connected

Dustovsky, 2022-03-03:
something I observed today (1.2.6)

In switzerland, we have regular 3 phase supply. in my case, I'm using 3x16A for the cars, 1x16A (max 3.6kw) for the Hyundai eKona and 2x16A for the Tesla (2x 3.6kW)

in the config.json, I edited the number of phases to "2" (because obviously, I have connected only two phases...)
Nevertheless, on the homescreen top righthand "charging*, I see only the power of one phase, although lateron the voltage of both phases is stated...

Everything works as designed, only the report/numbers on the TWCManager homepage are wrong and indicating only the power from one phase

example:
I set the power to 8A
The car get's 2x8A = 3.7kW
TWCManger prints only 1x8A = 1.85kW

altough the TWCManager homepage shows me the correct voltage for L1 and L2 (roughly 230V), and as well correctly, 0V for L3

What did I wrong?

TomsterID3, 2022-03-04:
Hi Dustovsky!
I´m facing the same problem here in Italy.
I use my TWC in 1phase connection during winter-month´s aand from spring to fall i would like to use it in 2phase connection to get all the extraJuice out of my Solar System.

I us a standalone PI with OpenWB to set the charging current in the TWCmanager.
OpenWB can read my Solaredge Inverter + Meter over Modbus TCP, and this works so much better than over API.

In the 1phase and 3phase mode it works perfectly, but not in 2phase, and this is because of the wrong powercalculation in the TWCmanager software.

So, oping for a solution on this !

THANKS for all your efford  ngardiner !!!

dehsgr, 2022-03-04:
I'm facing the same issue with three phases. There is only displayed one phase in the circle, regardless of three phases being used.

ChutneyMary, 2022-03-04:
@dehsgr - The number displayed 'in the circle' is the number of vehicles connected. The OP is talking about the formula for power being used while charging. ie. 5000 W.

dehsgr, 2022-03-05:
@ChutneyMary ok. Thanks for clarification. So I misinterpreted the circle. 😅 Sorry – my mistake.

ngardiner, 2022-03-05:
Hi All - I can confirm that the code didn't handle the case of 2-phase installs (and made some other assumptions about how wiring might be configured which I've relaxed). I've changed the logic in the development version 1.3.0 which should result in proper detection of 2-phase installs now and as a result should calculate wattage accurately, if you could give it a try and give me your feedback that would be great.

Dustovsky, 2022-03-16:
@ngardiner I did now an update, nevertheless it didn't change anything in the calculations... Honestly, I'm not sure if the update was successful at all :-/
Need to investigate later...

TomsterID3, 2022-03-20:
Made it to get the v1.3.0
I will test the TWC with 2 and 3 phases this week!

To get a successful update i hat to do those steps:

cd TWCManager
git checkout main -f
git pull
make install

Without the -f it didnt work!

TomsterID3, 2022-03-22:
I can say it works now!!
Succesfully charging my VW ID.3 on two phases with 6A gives me 2800 Watts in the TWCManger website!

Thanks to all of you!

ngardiner, 2022-03-23:
Wonderful, thanks for testing it @TomsterID3

MikeBishop, 2026-10-26:
Fixed in af780b8c (v1.3.0), which handles 2-phase supplies in phase detection, and confirmed working above. Closing.

closed 2026-10-26 by MikeBishop