ngardiner/TWCManager #422: Lifetime data looks wierd

KirkKirk, 2022-03-06:
The data on the Lifetime KWh shows lots of KWh which I believe it is a bug.

[image]

r3pTiLe1860, 2022-03-13:
Same for me (it‘s not new in current version (using main), already had this with 1.2.3

[image]

r3pTiLe1860, 2022-05-21:
Any idea on this? In my case it‘s a black signature edition one. Maybe this helps?

MikeBishop, 2022-05-23:
Obviously we can't do anything if the device is reporting the wrong data.  If it's reporting at an unusual scale, it should be possible to let you configure a multiplier.  (It *might* even be possible to infer the scale by looking at the change relative to charging, but let's start with the obvious.)

That almost looks like Wh rather than kWh.  Does it increase at a normal rate?  That is, if you charge at 10kW for 6 minutes, has it gone up 1 or 1000?  If 1, it's just reporting nonsense and you're out of luck; if 1000, a multiplier might be a reasonable fix.

dehsgr, 2022-05-23:
@MikeBishop I can confirm there has sth. change between v1.2.x and v1.3.x. After updating I could recognize the same (wrong) display. I think (but I did not confirm) that the kWh have to be divided by 1000. This would be a more possible value of charged kWh (in my case).

dehsgr, 2022-05-24:
I just monitored and can confirm the current value should be divided by 1000. I was loading with around 4,6 to 4,9 kW (green energy) and counter increased 1 kWh within around 8 minutes and 2 kWh within 19 minutes.

MikeBishop, 2022-05-24:
Did it go up by 1 and 2 respectively, or 1000 and 2000?

dehsgr, 2022-05-24:
It increased 1 by 1.

r3pTiLe1860, 2022-05-25:
> I just monitored and can confirm the current value should be divided by 1000. I was loading with around 4,6 to 4,9 kW (green energy) and counter increased 1 kWh within around 8 minutes and 2 kWh within 19 minutes.

I can **not** confirm this.

See the two screenshots i attached. I did the same test (10 minutes - 6KW) so i get 1kWh added as it is stated correctly, but also on the correct position. 

I checked my homecharging statistics at teslafi and it states i charged for about 2600kWh.

I found a screenshot from 05 last year (after i initially set TWCManager up) with the value 1769580. So nearly 1000kWh since last year which seams correct to me. 

so the start value had to be around 1767900 (1770500 - 2600)

That means in my case it‘s not about a factor, but maybe an offset? Or any other possible explanations about this value?

[image]
[image]

MikeBishop, 2022-06-06:
That's what I was afraid of.  If TWCManager is accurately reporting the value reported by the TWC and the TWC is just starting from / has acquired a nonsense number, our options on the software side seem limited.  A manual offset (but how would you know the correct value?) or simply suppressing the field in the UI seem the most likely.

r3pTiLe1860, 2022-06-06:
Yeah unfortunately. 

Personally i would appreciate an offset/hide function. So it‘s up to the user. In my case i can determine the approx offset. I mean currently it‘s wrong by 1.7mio kWh, so any value should be better than this :-)

I can‘t say on how many units this thing occurs but it seams that it‘s not unique. For sure it‘s not super important, but if there‘s an easy realisation way i would absolutely like it.

Eg: a option in the config file:
Default (not set): as TWC reports
-/+ value that is added / substracted from the TWC reported value
a specific value like -1 to just hide the reported value (but could be optionally)

wooter, 2022-11-27:
Maybe combine this with https://github.com/ngardiner/TWCManager/issues/289

My 2 TWC's were also reporting correct numbers until the summer. In august they were both wacky.

curlyel, 2023-10-15:
Sorry for digging out that old issue.
Is there a way to correct the values meanwhile through an offset or is it just a "wontfix" to live with?

My TWC's are both reporting nonsense lifetime counters as well:

[image]

Instead of showing weird counters, I'd prefer omiting the output..