ngardiner/TWCManager #428: twcman trying to start charging when not at home

dtiefnig, 2022-03-16:
I have a strange behavior here. TWCManager is trying to start charging my car, which is not at home. Home location and vehicle current location have been learned correctly by twcmanager and are displayed in the setting tab. They are about 20km apart. There is nothing connected to the TWC, still it seems twcmanager tries to start charging, also calling via Tesla API to my vehicle, which rejects to charge, as it is not connected to a charger. `"onlyChargeMultiCarsAtHome": true` is set in config.json.

Charge policy is "Non Scheduled Charging", which is okay, as it is after sunset and it is configured that way.

Not sure what the "Car stuck" message from Salve means and if it is maybe causing the behavior. There is nothing connected to the charger, I just double checked. :)

Log file says:
`
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (settings.chargeNowAmps [12]), condition (gt), value (0)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1647371054.5824907]), condition (gt), value (0)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1647371054.5824907]), condition (gt), value (1647454723.9416094)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Policy conditions were not matched.
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Policy conditions were not matched.
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (tm_hour [19]), condition (gte), value (6)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (tm_hour [19]), condition (lt), value (18)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Policy conditions were not matched.
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   13 Evaluating Policy match (settings.nonScheduledAction [1]), condition (lt), value (3)
Mar 16 19:18:43 charon twc-test[665]: ⛽ Policy   14 All policy conditions have matched. Policy chosen is Non Scheduled Charging
Mar 16 19:18:43 charon twc-test[665]: ⛽ Manager  20 Limiting charging to 6.00A.
Mar 16 19:18:43 charon twc-test[665]: ⛽ Manager  20 Charge when above 6A (minAmpsPerTWC).
Mar 16 19:18:43 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 19:18:43 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 19:18:44 charon twc-test[665]: ⛽ Manager  20 Slave TWC 7218: Delivered 1119 kWh, voltage per phase: (241, 236, 239).
Mar 16 19:18:44 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 19:18:44 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 19:18:44 charon twc-test[665]: 🚗 TeslaAPI 13 startOrStop is set to start
Mar 16 19:18:45 charon twc-test[665]: 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/149xxxxxxxxxx452/data_request/charge_state <Response [200]>
Mar 16 19:18:45 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 19:18:45 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 19:18:45 charon twc-test[665]: 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/149xxxxxxxxxx452/data_request/drive_state <Response [200]>
Mar 16 19:18:45 charon twc-test[665]: 🚗 TeslaAPI 19 Orion has departed
Mar 16 19:18:46 charon twc-test[665]: 🚗 TeslaAPI 13 Car API cmd charge_start <Response [200]>
Mar 16 19:18:46 charon twc-test[665]: 🚗 TeslaAPI 17 Orion: start charge response{'response': {'reason': 'disconnected', 'result': False}}
Mar 16 19:18:46 charon twc-test[665]: 🚗 TeslaAPI 20 ERROR "disconnected" when trying to start car charging via Tesla car API.  Will try again later.
Mar 16 19:18:46 charon twc-test[665]: If this error persists, please file an issue at https://github.com/ngardiner/TWCManager/ with a copy of this error.
Mar 16 19:18:46 charon twc-test[665]: 🚗 TeslaAPI 13 updateCarApiLastErrorTime() called due to Tesla API Error. Updating timestamp from 0 to 1647454726.1711724
Mar 16 19:18:46 charon twc-test[665]: ⛽ Slave    20 Car stuck when offered spikeAmpsToCancel6ALimit.  Offering 2 less.
Mar 16 19:18:46 charon twc-test[665]: ⛽ Slave    20 SHB 7218: 00 00.00/00.00A 0000 0000  M: 09 00.00/14.00A 0000 0000
Mar 16 19:18:46 charon twc-test[665]: ⛽ Slave    20 Slave power for TWCID 7218, status: 0
`

dtiefnig, 2022-03-16:
Car is at home now, same behavior.

dtiefnig, 2022-03-16:
After I connected the charger (with the car's charging limit set below the current battery state, so it doesn't charge), it did recognize the state:

Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 13 startOrStop is set to start
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/1492931557963452/data_request/charge_state <Response [200]>
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 13 Car API cmd charge_start <Response [200]>
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 17 Orion: start charge response{'response': {'reason': 'complete', 'result': False}}
Mar 16 21:45:57 charon twc-test[665]: 🚗 TeslaAPI 20 Orion is done charging or already trying to charge.  Stop asking to start charging.

But now the log is spammed with:

Mar 16 21:47:00 charon twc-test[665]: 🚗 TeslaAPI 13 startOrStop is set to start
Mar 16 21:47:00 charon twc-test[665]: 🚗 TeslaAPI 13 Don't charge Orion because vehicle.stopAskingToStartCharging == True
Mar 16 21:47:01 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 21:47:01 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 21:47:01 charon twc-test[665]: 🚗 TeslaAPI 13 startOrStop is set to start
Mar 16 21:47:01 charon twc-test[665]: 🚗 TeslaAPI 13 Don't charge Orion because vehicle.stopAskingToStartCharging == True
Mar 16 21:47:02 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 21:47:02 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True
Mar 16 21:47:02 charon twc-test[665]: 🚗 TeslaAPI 13 startOrStop is set to start
Mar 16 21:47:02 charon twc-test[665]: 🚗 TeslaAPI 13 Don't charge Orion because vehicle.stopAskingToStartCharging == True
Mar 16 21:47:03 charon twc-test[665]: 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mar 16 21:47:03 charon twc-test[665]: 🚗 TeslaAPI 13 car_api_available returning True

dtiefnig, 2022-03-16:
I did let it charge a little, and when it finished, it went into the "Don't charge" spam loop again.

dtiefnig, 2022-03-16:
Ah, as these messages are on log level 13, I guess it will be okay and just logging internal logic. There are no Tesla API calls sent, right?

dtiefnig, 2022-03-16:
So with connecting the charger, the problem stopped. This did also stop `Slave` logging the "Car stuck" messages.

dtiefnig, 2022-03-16:
After disconnecting the charger, `Slave` is logging the "Car stuck" messages again, but the Tesla API error did not reappear.

dtiefnig, 2022-03-16:
After restarting TWCManager, the Tesla API error reappeared as in my original comment above.

dtiefnig, 2022-03-16:
And after connecting/disconnecting the charger the problem is gone again. TWCManager still reports `Don't charge Orion because vehicle.stopAskingToStartCharging == True` once every second, which I guess prevents it from telling the car to start charging.

So I guess something is wrong in the state machine when TWCManager is (re)started. Wondering why nobody noticed this. I'm running current 1.3.0 btw. with my patches from #427 and #429.

dtiefnig, 2022-03-16:
Worked around the issue by adding `disconnected` to `Vehicle/TeslaAPI.py` line 708, as one of the states which tell TWCManager to stop charging the car. Still it should have never asked the car to start charging, at least while not at home.

I'll see if I can reproduce the issue tomorrow when the car is away again, or if it's only re-appearing while its at home.

dtiefnig, 2022-03-17:
Same behavior today with `Track Green Energy` policy. Car is about 20 km away from home, and TWCManager tells it to start charging as soon as there is enough green power to start. So similar to #423 but home locations is perfectly fine.

When the car is sleeping though when TWCManager starts up, I can see its location is not marked on the map on the settings web page until TWCManager was able to wake the car. Maybe its a logic error, car location is only updated after charging command has been sent?

dtiefnig, 2022-03-17:
Found the problem. `car_api_charge(charge)` in `Vehicle/TeslaAPI.py` is only checking whether a car is at home if `self.config["config"]["onlyChargeMultiCarsAtHome"]` **AND** `self.getVehicleCount() > 1`. [lines 615/616](https://github.com/ngardiner/TWCManager/blob/9adc31987026971c7916666f5864972f341dcb9e/lib/TWCManager/Vehicle/TeslaAPI.py#L615)

This does make sense in regards to the config option (as it explicitly refers to "MultiCars") but this causes a car to always be started/stopped if it is the only car in a Tesla account. I don't think there's a valid use-case for this, is there? I think "single car charging" was accidentally broken by cdragon when he implemented the multi car (not) at home charging on Aug 24 2018. Before that change, the function did check for every car (even if there's only one) whether it is at home: [0a15cf0](https://github.com/ngardiner/TWCManager/commit/0a15cf03acd040716038627865717cfcae43a659)

So my suggestion would be to obey `onlyChargeMultiCarsAtHome` even if **NOT** `self.getVehicleCount() > 1`. Probably renaming the config option or introducing a new one (at least for a transition or permanently for backwards compatibility) would be a good idea. But just removing [line 616 in Vehicle/TeslaAPI.py](https://github.com/ngardiner/TWCManager/blob/9adc31987026971c7916666f5864972f341dcb9e/lib/TWCManager/Vehicle/TeslaAPI.py#L616) would do the trick.

Or is there any reason why we would want to treat a car differently if it is the only one in an account?

MikeBishop, 2022-03-21:
At one point, the assumption was that sending "start" commands is harmless if the car isn't connected to the charger, and "stop" commands only get sent when a car is actually charging.  If you only have one car, the GPS doesn't matter -- if a car is charging, safe to assume it's yours.  GPS wasn't always reliable, so some people would rather ignore it.

I doubt that logic holds these days, and personally, I'd like to just eliminate the conditional entirely.  I'd be okay with eliminating the multi-car condition, at least, and just leave the option as the ability to ignore GPS.  (Note that other pieces that use GPS already ignore this setting.)  @ngardiner, what do you think?

ngardiner, 2022-03-21:
I'm in complete agreement, one less config setting and less complexity equal a win/win in my books.

dtiefnig, 2022-03-21:
If you think we don't cause issues with older cars, just removing the config option and `if` test is sure the easiest solution.

PR #435 (2022-04-18): remove onlyChargeMultiCarsAtHome
Removed the test for onlyChargeMultiCarsAtHome from `TeslaAPI.py` and removed the config option from `etc/twcmanager/.testconfig.json`, `etc/twcmanager/config.json`, `etc/twcmanager/demo.json` and `tests/API/test_getConfig.py`.
 Fixes #428
commit d4a993ef (2022-04-18): remove onlyChargeMultiCarsAtHome check

Do not check "onlyChargeMultiCarsAtHome" config option before checking
whether a car is at home or not. Also removes check if there is more than 1
car enrolled in the account. Fixes #428
closed 2022-04-18 by ngardiner