ngardiner/TWCManager #433: UnicodeEncodeError

Heptagon404, 2022-03-20:
Hello together, 

I did a reinstall of TWCManager from dracoventions to the ngardiner version. 
Now I get this errors when starting 
sudo -u twcmanager python -m TWCManager (also without sudo)

--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1082, in emit
    stream.write(msg + self.terminator)
  File "/usr/lib/python3.9/encodings/iso8859_15.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u26fd' in position 23: character maps to <undefined>
Call stack:
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 619, in <module>
    master.registerModule(
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 931, in registerModule
    logger.log(
Message: 'Registered module %s'
Arguments: ('ConsoleLogging',)
--- Logging error ---

<loop>

With loglevel 11 I get this in debug.log (without xxxx)
^[[33m23:12:56^[[0m ^[[32murllib3.co^[[0m ^[[36m10^[[0m Starting new HTTPS connection (1): api.sunrise-sunset.org:443
^[[33m23:12:58^[[0m ^[[32murllib3.co^[[0m ^[[36m10^[[0m https://api.sunrise-sunset.org:443 "GET /json?lat=47.xxxxx&lng=9.xxxxx&formatted=0&date=2022-3-20 HTTP/1.1" 200 None

I spend quite some time, but was not able to fix this issue.
Any help is very welcome. 

Cheers,

Thomas

MikeBishop, 2022-03-21:
What platform are you on?  It looks like Python is choking trying to output Unicode characters, but I'm surprised that would be a problem on any modern system.

Heptagon404, 2022-03-27:
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

Heptagon404, 2022-04-04:
Reinstall of the OS fixed this issue.

closed 2022-04-04 by Heptagon404