ngardiner/TWCManager #437: single phase charging on three phase TWC

dtiefnig, 2022-03-21:
I have a three phase power setup for my TWC and today connected a car which is only charging via one phase. TWCManager is not able to handle this correctly, as it is not able to determine which/how many phases a TWC/car is using to charge. The slave just reports 16 Ampere in use, and TWCManager calculates it is using 11 kW. But actually it is only 3.7 kW.

I don't think this can be easily fixed, as it seems that slave is not reporting usage per phase. Or maybe we just did not yet find the right message to send to the slave to make it respond with Amperes per phase?

We could try to derive actual usage from reported lifetimekWh, as the TWC is calculating that correctly. But that's an ugly hack which would not even work well, as it can take up to 45 minutes (6A at 230V) to increase by 1 kWh.

Do you see any options to detect single vs. multi phase usage in such case?

ngardiner, 2022-03-21:
It may be possible to detect based on the current drawn vs offered, however it's such a corner case - we have only had one report of it previously. Perhaps if you have access to the vehicle you could take a look at all of the relevant readings for us, and see if any of them might identify to TWCManager that such a vehicle is charging. I'll likely never have such a vehicle and don't even have a 3 phase install so ultimately I don't foresee myself proposing a fix.

Even still - compensating for it may be a whole other matter. We're already constrained as to the maximum current for 3 phase, and offering over any circuit limit would be dangerous even if we were quite sure that the vehicle was single phase only, so the reality is that it's pretty much just cosmetic and any fix has to be weighed up in terms of accuracy - if it has the potential to break calculations for others but simply fixes a cosmetic reading for situations where a 3 phase install (already only a proportion of installs) has a single phase vehicle only (which is a very small subset of vehicles as well) - is it potentially more trouble than it is worth?

Also these same vehicles tend not to respond to CAN bus queries either, so we can't just pull VIN and identify them that way.

dtiefnig, 2022-03-22:
It is a minor issue indeed, even though not purely cosmetic, as green power charging would not work correctly, because calculation is based on 3 phases. So when 1.4 kW of green power is available, we could start charging with 6A at 230V, but it would only start at ~4.1 kW, i.e. when it could provide 6A on three phases. At 4.1kW in turn, it would charge with 6A while it could use up to 18A theoretically.

Still implementing any hacks may not be worth the trouble. Three phase installs should be the standard throughout Europe at least, but most cars with larger batteries will support three-phase charging I guess. The car I had charging via 1 phase only is hybrid electric.

I'll see if I can collect more information.

ngardiner, 2022-03-23:
> It is a minor issue indeed, even though not purely cosmetic, as green power charging would not work correctly, because calculation is based on 3 phases. So when 1.4 kW of green power is available, [...]

The issue is that we speak only to the TWC, instructing the TWC to charge at a certain rate across all 3 phases. If the vehicle only then utilizes one phase, they get only the proportion of energy associated with that phase.

We can't instruct the TWC to increase the rate for a single phase and even if we could it's not like electrical systems can divert energy from other phases so the vehicle is still limited in terms of circuit capability, and your example above wouldn't work even if we tried it (compensating for the single phase draw by multiplying the offering to 18A) because 16A is the maximum for 3-phase TWCs, plus it's still limited by the overall circuit capacity so if there's less than 16A available, that free capacity dwindles. 

To a large degree as well it depends on the green energy system. Is it feeding all 3 phases, and how is the energy feed back into the grid metered?  

So we get left in a circle of compensating for a limitation of the TWC itself (remember that single phase vehicles will have this issue regardless of which TWC they connect to, TWCManager-managed or hard set, they'll always draw only one phase worth of the total circuit capability) but with only so much rope from which to draw. I've looked at this before and couldn't find a solution that was elegant without a lot of fudging numbers and a lot of questions about whether it would actually be beneficial, but you may find otherwise.

One final option to consider, and something I'd happily add to TWCManager if it was needed, is that others have turned off Phases 2 & 3 to get more granular control over charge rates in lower generation conditions. If someone had a single phase vehicle and wanted to switch between, the feeback from those users is that dynamic switching off of the phases has worked seamlessly however I can't test it myself.  A couple of relays on Phase 2 & 3 and the software could just switch them off and revert to Single Phase operation (again... untested/unimplemented). Overall that's how I'd approach it if I had the same issue.

dtiefnig, 2022-03-24:
The 3-phase TWC's (Gen 2 at least) are designed for (3x) 32A. I've limited my installation to 3x 16A though as it doesn't make a lot of sense to charge with more than 11 kW at home anyway.

You're right of course with Green Energy charging, I was completely mistaken here. The converter feeds in its power on all three phases, and we will utilize only one of the phases when a car is doing 1-phase charging. The TWC would have to do some tricks to "transform" 3-P to 1-P power, which it pretty sure can not do.

Energy transfer to the grid is metered on per phase basis here, so if one of the phases draws 1A, one pushes 2A out and one pushes 3A out, I'll have to pay for 1A delivered to me while getting paid for 5A at the same time. Would be a nice feature for the converter btw. to be able to compensate this by asymmetric feeding, which we then could also use to improve 1-P charging. Not our business though. :)

Shutting off phases dynamically is an interesting approach. Would not have expected this to work and it can be dangerous, if the device attached is not capable of handling it. I'd have expected the TWC to go into a failure state and not deliver any power until power supply is fully restored (and probably a manual reset).

It's a different topic though, a prerequisite step would be to determine at all whether a car is charging 1-P or 3-P. E.g. with a smart meter on the lines running to the TWC that should be possible, if we can't get the information from the TWC.

It would be really nice to get the Ampere readings from the TWC still, also to correctly calculate the power usage, not just in 1-P but also in 3-P charging situations, where voltage and current vary a little from phase to phase.

dtiefnig, 2022-03-24:
I'm attaching a full debug log of the first 2 minutes of a 1-P charging session here, just in case it is of any value for someone looking into this in the future: [docker-twc-test-1p-charge.log](https://github.com/ngardiner/TWCManager/files/8340953/docker-twc-test-1p-charge.log)

ngardiner, 2022-03-24:
> The 3-phase TWC's (Gen 2 at least) are designed for (3x) 32A. I've limited my installation to 3x 16A though as it doesn't make a lot of sense to charge with more than 11 kW at home anyway.

You're right - I'd conflated the limitation of the onboard 11kW charger in Model 3s and Ys with the TWC maximum, that said it would require people who have 3-phase supply to cable and size breakers for a higher current than their vehicles are likely to support (unless they have an S or X) so the limitation is still going to be inherent in many installs simply based on the cost being effectively doubled from a cabling and breaker perspective to support 32A when the Model X and Y won't support it.

> Shutting off phases dynamically is an interesting approach. Would not have expected this to work and it can be dangerous, if the device attached is not capable of handling it. I'd have expected the [...]

Sure, and with all the usual disclaimers since we are dealing with pushing the hardware beyond its documented capabilities, some brave souls have been successful however we can't be sure that there's no inherent risk to this approach.

At the same time, individual phase supply failures happen and the unit would have to respond to those, it may be that it was designed with this in mind hence the ability for the unit to cope, but that's speculation on my part.

> It's a different topic though, a prerequisite step would be to determine at all whether a car is charging 1-P or 3-P. E.g. with a smart meter on the lines running to the TWC that should be possible, if we can't get the information from the TWC.

There still could be corner cases, such as a supply failure on one or more phases making it look like the vehicle is only drawing from one phase (although voltage measurements would help to distinguish this), but how TWCManager responds to that information would determine whether it really matters.

Unfortunately there's no standard protocol to use here to measure this. It would need to be coded up, in whatever form makes sense, it would require the additional hardware, but it is doable.

MikeBishop, 2022-03-24:
> I'm attaching a full debug log of the first 2 minutes of a 1-P charging session here, just in case it is of any value for someone looking into this in the future: [docker-twc-test-1p-charge.log](https://github.com/ngardiner/TWCManager/files/8340953/docker-twc-test-1p-charge.log)

Just FYI, that log contains your unredacted latitude/longitude on line 523.

dtiefnig, 2022-03-24:
Yeah, thanks, I noticed that, forgot to reduce the precision at least. But I'm not hard to find anyway, as due to being an entrepreneur in my country, my address is published in many public registers. So I don't mind really.

Flitzer42, 2022-05-07:
Some comments...
Single phase charging: Model 3 supports higher currents when only charging on a single phase. I use it up to 25A to have a wider range of charging power (1.4 ... 5.7 kW). Fortunately my meter allows to exchange the power between phases via the grid without aditional costs. But there are legallimitation regarding the inbalance between the phases.
Swithing phases: OpenWB allows to switch phases and as I know, Teslas are fine with it, whereas other vehicles get permanently damaged. So I would not recommend to do so during charging. As i know, it is necessary to interrupt the CP signals when changing the phases while the vehicle is connected otherwise, it will promt with a faillure. TWC does not allow to controll the CP signal.

dtiefnig, 2022-08-16:
I'm closing this issue, as there's not much we can do now (or in the future, likely).

closed 2022-08-16 by dtiefnig