ngardiner/TWCManager #438: Apply Charge Limit Error

eszych, 2022-03-23:
I am getting this error in the Log-File:
```
  File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 255, in background_tasks_thread
    carapi.applyChargeLimit(limit=task["limit"])
  File "/usr/src/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 868, in applyChargeLimit
    located = vehicle.update_location()
  File "/usr/src/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1391, in update_location
    self.atHome = self.carapi.is_location_home(self.lat, self.lon)
  File "/usr/src/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 548, in is_location_home
    abs(self.master.getHomeLatLon()[0] - lat) > 0.0289
TypeError: unsupported operand type(s) for -: 'str' and 'float'
, occurred when processing background task

```
Version 1.3.0 - updated today

MikeBishop, 2022-03-24:
If you look in your `settings.json`, is your home location stored as a string or a number?  We should probably coerce that to a number on load anyway, but it looks like that's the immediate cause.

MikeBishop, 2022-03-24:
Assuming that's the issue, can you check that #439 addresses this for you?  Would you be willing to share what the string in your `settings.json` looks like?  (No need for the actual value, of course.)

closed 2022-03-24 by ngardiner

eszych, 2022-03-24:
here is the part from the settings.json:
`{"chargeNowAmps": 0, "chargeStopMode": 1, "chargeNowTimeEnd": 0, "homeLat": "50.118871", "homeLon": "8.494693", "hourResumeTrackGreenEnergy": 16, "kWhDelivered": 119,
`
so, I guess it is a string as it is stored in double quotes...
I am not a programmer though and might be wrong here...

And WOW! Thanks for the fast response!
This is impressive piece of work that you all have delivered!

MikeBishop, 2022-03-25:
Yeah, coercing it to a float should fix it for you, then.  But I'd really like to find the bug that's causing us to write out strings instead of numbers....

Happy to help!  It was a simple fix, and I happened to see it when I had a free moment.  Some other requests are not so simple and don't get addressed anywhere near as quickly.  😖  (And my apologies to those people!)

PR #439 (2022-03-24): Coerce lat/lon into floats on retrieval
Fixes #438 by forcing the value from `settings.json` to be a float.  Should be a no-op, but it appears some people are getting these written as strings occasionally.