ngardiner/TWCManager #440: bore installation after update to 1.2.4/1.2.6 (hopefully just a noob issue)

danielsan1, 2022-03-27:
Hi @ngardiner (and community :)

I just tried to update my 1.2.3 installation to 1.2.4 (and after that failed also to 1.2.6)
After the installation I'm only able to start the "old interface" with http://192.168.188.102 if I use :8080 i get a "can not connect".

systemctl status twcmanager.service says: 

● twcmanager.service - Tesla Wall Charger Manager
   Loaded: loaded (/etc/systemd/system/twcmanager.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-03-27 16:41:50 CEST; 1min 22s ago
  Process: 739 ExecStart=/usr/bin/python3 -u -m TWCManager (code=exited, status=1/FAILURE)
 Main PID: 739 (code=exited, status=1/FAILURE)

Mär 27 16:41:50 twcmanager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Mär 27 16:41:50 twcmanager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Mär 27 16:41:50 twcmanager systemd[1]: Stopped Tesla Wall Charger Manager.
Mär 27 16:41:50 twcmanager systemd[1]: twcmanager.service: Start request repeated too quickly.
Mär 27 16:41:50 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 27 16:41:50 twcmanager systemd[1]: Failed to start Tesla Wall Charger Manager.
Mär 27 16:42:57 twcmanager systemd[1]: twcmanager.service: Start request repeated too quickly.
Mär 27 16:42:57 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 27 16:42:57 twcmanager systemd[1]: Failed to start Tesla Wall Charger Manager.

and journalctl -f says:

-- Logs begin at Sun 2022-03-27 16:41:34 CEST. --
Mär 27 16:44:40 twcmanager python3[883]:     import yaml
Mär 27 16:44:40 twcmanager python3[883]: ModuleNotFoundError: No module named 'yaml'
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Main process exited, code=exited, status=1/FAILURE
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Mär 27 16:44:40 twcmanager systemd[1]: Stopped Tesla Wall Charger Manager.
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Start request repeated too quickly.
Mär 27 16:44:40 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 27 16:44:40 twcmanager systemd[1]: Failed to start Tesla Wall Charger Manager.

As I'm not a Linux pro - is it maybe something easy to fix? Or do I need to do a full reinstall? :(
I did the manual install and upgraded over the last months and years multiple times without issues using
git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout v1.2.4
sudo make install

Thank you very much in advance!

ngardiner, 2022-03-28:
Hi @danielsan1,

It looks like something is failing during the ```make install```, do you get any errors when you run it?

My recommendation would be to go to v1.2.6 because there were some issues with the way we handled installation in v1.2.4 and some changes to python setuptools, however if both are failing it seems like the issue might be elsewhere.

danielsan1, 2022-03-28:
Thanks @ngardiner for your fast reply! No errors during the installation except "ModuleNotFoundError: No module named 'yaml'" 

pi@twcmanager:~ $ cd TWCManager
pi@twcmanager:~/TWCManager $ git checkout v1.2.6
HEAD ist jetzt bei 34d0545 fixup: Format Python code with Black
pi@twcmanager:~/TWCManager $ sudo make install
sudo apt-get update
OK:1 http://archive.raspberrypi.org/debian buster InRelease
Holen:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]
Holen:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13,0 MB]
Es wurden 13,0 MB in 10 s geholt (1.343 kB/s).                                                                                                                   
Paketlisten werden gelesen... Fertig
sudo apt-get install -y git libffi-dev libpq-dev libssl-dev
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
git ist schon die neueste Version (1:2.20.1-2+deb10u3).
libffi-dev ist schon die neueste Version (3.2.1-9).
libpq-dev ist schon die neueste Version (11.14-0+deb10u1).
libssl-dev ist schon die neueste Version (1.1.1n-0+deb10u1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
# Create twcmanager user and group
sudo useradd -U -M twcmanager 2>/dev/null; exit 0
sudo usermod -a -G dialout twcmanager
# Create configuration directory
sudo mkdir -p /etc/twcmanager
sudo chown twcmanager:twcmanager /etc/twcmanager -R
sudo chmod 755 /etc/twcmanager -R
pi@twcmanager:~/TWCManager $ sudo systemctl start twcmanager
pi@twcmanager:~/TWCManager $ journalctl -f
-- Logs begin at Sun 2022-03-27 16:41:34 CEST. --
Mär 28 07:59:18 twcmanager python3[13755]:     import yaml
Mär 28 07:59:18 twcmanager python3[13755]: ModuleNotFoundError: No module named 'yaml'
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Main process exited, code=exited, status=1/FAILURE
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Mär 28 07:59:18 twcmanager systemd[1]: Stopped Tesla Wall Charger Manager.
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Start request repeated too quickly.
Mär 28 07:59:18 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Mär 28 07:59:18 twcmanager systemd[1]: Failed to start Tesla Wall Charger Manager.

Any ideas?

What would be the easiest way to do a full reinstall? And can I keep my config file?

Thanks!

dehsgr, 2022-03-28:
@danielsan1 please try the following command:
`pip3 install pyyaml`

Python YAML seems to be missing when moving from 1.2.4 to 1.2.6. That solved it for me.

danielsan1, 2022-03-28:
**That solved it! 👍 Thank you very much @dehsgr !** 

Just one more thing: Now I see at the top of the web interface:
"We experienced an error trying to save your settings. This means that while your current settings will be retained for the current session, they will not be saved if TWCManager exits. Please check the permissions on /etc/twcmanager/settings.json and try saving your settings again."

Do you know the correct permissions? As far as I know I never changed them/touched the setting file...
pi@twcmanager:/etc/twcmanager $ ls -l
insgesamt 36
-rwxr-xr-x 1 twcmanager twcmanager 27096 Aug 28  2021 config.json
-rwxr-xr-x 1 twcmanager twcmanager  5734 Mär 27 14:30 settings.json

dehsgr, 2022-03-28:
You’re welcome. :-) 

Is the executing user really „twcmanager“? Normally the file is owned by pi:pi…

danielsan1, 2022-03-28:
Oh that's strange indeed... I fixed it with chown! Thanks again :)

closed 2022-03-28 by danielsan1