ngardiner/TWCManager #445: Questions regarding “on/off/on/off” charging in cloudy conditions

danielsan1, 2022-04-10:
Hi! 

As it is typical “April weather” today I experience heavy “on/off charging” and asked myself if that is in any way harmful to the charger or relays. 
(With “on/off/on” I mean the one moment it’s super sunny and I’m charging with 10A 3 phases. The other moment it’s very cloudy and power drops below 5A and TWCManager stops charging. This repeats in very short intervals)

1. Does this might be harmful to any component if it’s going on/off in super short intervals?
2. I heard from friends using other solar charging setups that their system can even go down to 1 phase and charge the lowest at 1.x kw - is that possible with TWCManager?
3. If I set the config option to keep on charging at 5A even solar power will be below that threshold, is there any option to manually stop charging if I see the weather conditions change so it makes sense to stop charging least for an hour or so. (If I use “stop charging” in the Tesla app it will anyway resume after a few seconds)

Thanks for some clarification and some advice :)

danielsan1, 2022-04-11:
Thank you @MikeBishop ! I wasn’t aware of the startStopDelay option. I’ll definitely have a look at it. 
I already use the greenEnergyFlexAmps option as I’m afraid that the short interval stops are harmful to the changing electronic. But do you have any knowledge/experience on that? Is it really harmful to the charger or relays? What would be a „save“ delay?
Thanks again!

MikeBishop, 2022-04-11:
It defaults to one minute; I increase mine to five minutes.

Theoretically, any contactor has a limited number of cycles it will survive, so by definition fewer cycles on it means a longer life.  On the other hand, imagine a car that isn't using TWCManager -- the contactor will engage basically any time you return home and plug it in.  Using TWCManager and _occasionally_ having a day like that is probably no worse that someone who makes multiple short trips a day and recharges 2-3% after each one, a usage pattern they presumably designed to accommodate.

I've added a number of tweaks in an effort to avoid constantly starting and stopping on cloudy days, but as long as they're not every day, I doubt it's going to seriously impact the lifespan of the TWC.

danielsan1, 2022-04-11:
Thank you very much for the detailed explanation!
I’ll configure the delay to 5 min and observe the results. This will probably reduce start/stop charging sessions to a “healthy” level.

closed 2022-04-11 by danielsan1