ngardiner/TWCManager #446: Pilot Signal and TWCManager

ChutneyMary, 2022-04-12:
I'm experiencing a strange intermittent problem where the charging cable won't lock in sometimes. I believe a 'Pilot Signal' needs to be present when the charge cable is inserted to trigger the locking of the charge port.

My issue might have no relevance to TWCManager and might be a simple hardware failure in the Pilot Signal circuit. Notwithstanding, does TWCManager have any input or relevance to the Pilot Signal being present?

MikeBishop, 2022-04-18:
Not in the typical configuration, but it's possible.  The TWC will go into an error mode (red flashing light on the front, no pilot signal) if it's set to slave mode and not detecting a master TWC (which is TWCManager in this case).  So if you have TWCManager configured, but the process is stopped (or serial communication is interrupted), the TWC won't do anything.  Likewise, if TWCManager's selected method to stop charging is to stop responding to slaves, you'll see that behavior when TWCManager doesn't want to offer any current.

Normally, though, TWCManager tells each device to offer the minimum amount of power supported (5-6A) and uses the Tesla API to stop charging when not desired.

If it's intermittent and a red light, I'd check the TWCManager output for crashes or loss of communication.  If the TWC itself isn't displaying an error state, I'd guess it's probably the TWC hardware itself.

ChutneyMary, 2022-04-18:
Thanks. I've since been to the local Tesla repair centre and they have some ability to do a remote diagnosis of the TWC (I don't know how...?)
They have deemed it to be a problem with the Pilot Signal and have given me a replacement TWC.

closed 2022-04-18 by ChutneyMary

ngardiner, 2022-04-18:
Thanks for the update @ChutneyMary