ngardiner/TWCManager #450: Disconnected Install Guidance

MikeBishop, 2022-04-26:
Now that Tesla API control is enabled, it's entirely possible to run TWCManager and take _almost_ full advantage of it without physically connecting it to the TWC.  At a minimum, I think we should have instructions oriented to people who just want to run TWCManager in Docker with no serial connection and exclusively use the Tesla API to control the charge limit.

Given that we can't control Gen3 TWCs anyway, that might even need to be the default configuration for the project moving forward.  Those of us with Gen2 TWCs are going to be rarer and rarer as time goes on.

MikeBishop, 2022-04-26:
I think the changes we would need are:
- Ability to run with no serial interface
- "Virtual" TWC Slave instances for each car on the Tesla API, only if serial is not used (by default):
    - Only for cars that are at home

We'd need some logic to dedupe those virtual instances if someone were using a hybrid installation, but that could come later.

drexcia, 2022-04-30:
Have you seen Charge HQ? They're running in beta at the moment. It's all completely cloud based, no hardware required. Aside from the Tesla API token all l I had to do was add an account to my Fronius Solarweb instance so that they could read the generation/consumption data. I do prefer TWCManager though, quicker to read my generation/consumption info and start/stop charging. 

I'm sure that Jay over at Charge HQ would be willing to have a chat?

https://chargehq.net/

MikeBishop, 2026-10-26:
v1.4.0 makes a disconnected install possible: the RS485 interface unloads cleanly when no port is configured (#612), Tesla vehicles without a serving TWC can be charge-rate controlled via the Fleet API (6f075db0), and Gen3 units can be managed via Neurio emulation (`docs/modules/Controller_Gen3TWCs.md`). Could you re-verify whether that covers your setup? We don't yet have a write-up for an API-only install.

closed 2026-10-29 by MikeBishop