ngardiner/TWCManager #451: Charge limit no longer changing with policies

alexeiw123, 2022-04-27:
Recently I've noticed my car charging during scheduled time when SoC is above 50%.
My configuration in /etc/twcmanager/config.json is as follows:

```
        "chargeNowLimit": -1,
        "greenEnergyLimit": 90,
        "scheduledLimit": 50,
        #"nonScheduledLimit": -1,
```

This was working fine with no updates or config changes taking place, so I assume this must be TeslaAPI related. I am able to start/stop charging OK so I think the API connection is OK. TWCManager is v1.2.6.

If I manually adjust the charge limit from the app, TWC Manager doesn't change it back to 50% or 90% in either scheduled or green energy policy times.

If logs are helpful, I'll just need a but of a noob guide on how I go about providing them.

MikeBishop, 2022-04-28:
If you can stop charging successfully, I agree, it sounds like the API piece is correct.  TWCManager is very cautious not to change the SoC for a car that it isn't sure is home; it's possible that your home location has gotten messed up in `settings.json`.  Two possibilities: you could either remove the location from that file, restart TWCManager, and let it be rediscovered; or you could see if the change in #306 helps -- it will automatically reset the home location to the location of a car we see charging.

To test #306, you would `git checkout location_frequency` and restart TWCManager.

alexeiw123, 2022-04-29:
> it's possible that your home location has gotten messed up in settings.json

I think you're right. I grabbed the lat/lon from `settings.json` and threw it in to google - it shows as being a good 600m or so from my house.

I've removed the location fields from `settings.json` and restarted the service. My wife is out driving around for now so I will check back on it later.

I decided no to test #306 simply as I'd not be confident in bug testing or reverting those changes if it didn't work as expected.

alexeiw123, 2022-04-30:
> If you can stop charging successfully, I agree, it sounds like the API piece is correct. TWCManager is very cautious not to change the SoC for a car that it isn't sure is home; it's possible that [...]

Thanks Mike, this advice helped me to fix the issue.

I ended up clearing the contents of `settings.json`, copy across the vehicle charge history and manually configuring the other settings again. All working correctly once more.

I noticed in my old `settings.json` there was a saved setting `"chargeLimits": {"198532562955": [90, 90], "1492931160953009": [null, -1]}`. I'm not sure what UI configurable setting this refers to, could it have had something to do with it?

MikeBishop, 2022-05-02:
That element is used in knowing whether to update (and how to restore) the charge limits to match the policy.  The key is the ID for each vehicle; the first number is the vehicle's "normal" charge limit (so we can restore it when the vehicle leaves or a policy no longer applies one); the second number is the most recent charge limit applied by policy to the vehicle.

I'm a bit surprised that you could get a `null` in that spot.  It's entirely possible that the actual source of the problem was the unexpected value.

alexeiw123, 2022-05-02:
Thanks Mike, fixed for my install at this point regardless of cause.

leeliu, 2022-05-28:
I'm having a similar problem but can't seem to fix. My config is similar:
```
        "chargeNowLimit": 100,
        "greenEnergyLimit": 90,
        "scheduledLimit": 40,
        #"nonScheduledLimit": -1,
```
But it always seems to restore the charge limit to whatever's set in the settings.json, currently 80%. It restores to 80% at midnight (my scheduled start time, which is wrong...should be 40%) or whenever the car departs home (which is correct at 80%). I've checked that my lat/lng is set correctly. I've also tried removing the chargeLimits section of the settings.json but didn't seem to have much effect. I'm on v1.2.6.

alexeiw123, 2022-05-28:
I'm pretty sure that the minimum value you can set is 50. Try changing your 40 to a 50 and see if you're still seeing that behaviour.

leeliu, 2022-05-28:
Ooh...interesting...lemme test that tonight. Thanks!

MikeBishop, 2022-05-28:
@ngardiner is correct -- anything less than 50 or greater than 100 is equivalent to `-1` -- return to the original limit.

leeliu, 2022-05-28:
Yup that was it. The issues I do believe started when I changed the 50 to 40 a couple weeks ago. Just didn't see the note about 50-100 being the valid range (probably saw it initially years ago). Anyway thanks @alexeiw123 for helping troubleshoot that!

ngardiner, 2022-08-26:
I might close this one off now, but with a note: I've noted that there's been a few (more than a few, to be honest) instances of what looks to be home lat/lon related issues recently, I haven't pinpointed the reason, but I'm going to spend some time today working on some automated tests for this to see if I can pin it down somewhere. There is a chance that the location moving ~600m away is related to these issues but it's hard to say without further investigation.

closed 2022-08-26 by ngardiner