ngardiner/TWCManager #456: Not able to start, please help

kimstor1435, 2022-06-23:
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/kimstoroy/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/home/kimstoroy/TWCManager/lib/TWCManager/TWCManager.py", line 559, in <module>
    modinstance = modclassref(master)
  File "/home/kimstoroy/TWCManager/lib/TWCManager/Interface/RS485.py", line 52, in __init__
    self.ser = self.serial.serial_for_url(self.port, self.baud, timeout=0)
  File "/usr/local/lib/python3.9/dist-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

MikeBishop, 2022-06-23:
It isn't able to find your USB serial connector to talk to the TWC.  Do you have that installed?

kimstor1435, 2022-06-23:
Just installed the adapter, now it`s running :)

closed 2022-06-23 by kimstor1435