ngardiner/TWCManager #457: TWCManager stopping charge on Mobile Charger

ChutneyMary, 2022-06-30:
My installation has a Tesla Wall Charger with TWCManager installed in a front carport. This services the first Tesla. A second Tesla is in the rear garage with a Tesla Mobile Charger.
I was stumped why the second Tesla would sometimes stop charging after a few minutes when connected to the Tesla Mobile Charger. I resolved it by selecting 'Charge Now' in TWCManager to a value that provided an 'Offered' value greater than zero.
I'm guessing that the Tesla API is sending the 'stop charging' command to the second Tesla because the Solar Surplus hasn't reached the minimum amount (4.5kW). The second Tesla is only using single phase (2.4kW) with the Mobile Charger.
I don't know if there's a work-around to program to use the VIN when connected to the 'main' TWC managed wall charger. Then the Solar Surplus trigger will only be relevant to the vehicle plugged in to the main unit.
Otherwise this is probably an edge-case that can be managed knowing the Mobile Charger is affected by TWCManager.

MikeBishop, 2022-07-01:
Latitude and longitude is in `settings.json` and can be set from the Settings page, either using a map control or by pulling the current location of a vehicle.  The config value `atHomeRadius` can be used to control the sensitivity, and defaults to about two miles.  However, be aware that the default is so large because reported location isn't always precise (particularly in a garage); you might get some false positives / false negatives if you make it too narrow.

ChutneyMary, 2022-09-13:
I've attempted to experiment with the `atHomeRadius` to stop the influence of the TWCManager on a second Tesla parked in the rear garage (the first Tesla is parked in the front carport with the Tesla Wall Charger - separation is about 30 metres).
Results have been unsatisfactory. I suspect the GPS position drifts with the metal roof over each vehicle and I can't reliably exclude one vehicle from the atHome location.
I've since installed a Tesla Wall Charger in the rear garage (as opposed to using the Mobile Charger). I intend to fit this with a TWCManager if I can independantly control each vehicle's charging behaviour.
Is there some way to include a specific VIN with the Start/Stop API command to target a specific vehicle? My experience with `atHomeRadius` seems to suggest that all vehicles in an account are broadcast to?

MikeBishop, 2022-09-14:
Currently, no -- it starts/stops all vehicles on the account which are marked as being "at home."  Two things that might be easy to add, off the top of my head, but both require code changes:

- We could add an "ignore list" of VINs which TWCManager disregards from your Tesla account
- TWCs which report the VIN of the connected vehicle could send targeted stop/start commands to that vehicle and leave all others alone; not all provide this, so the current behavior would need to remain as a fallback.

There's been discussion about network-based modes of operation, either using the API to control a charging vehicle without control of the TWC or having two TWCManager instances that collaborate with each other, but both of those would be much larger projects that haven't been undertaken yet.

ChutneyMary, 2022-09-14:
Thanks for the information. I'd like to leave this issue open with the hope that a VIN "ignore list" could be implemented at some point. I think the occurrence of two home charging Tesla's will be an increasing use-case as ICE vehicles are swapped out.

ChutneyMary, 2022-09-17:
As an interim measure, I've settled on designing my own logic control using the Tesla API and my home automation system. Charge-Rate, Charge-Stop and Charge-Start are signalled with Node-Red  (node-red-contrib-tesla). I still need the data flow from the Raspberry Pi/TWCManager to assist with the control loop.
How can I have TWCManager set to a 'dumb' mode whereby it has no influence on the available amps? I just need it to continue to report various metrics whilst always providing 16amps available. I also don't want it sending any Start/Stop API commands.

MikeBishop, 2022-09-22:
You could disable the solar portion, remove your Tesla account, and set the Non-Scheduled behavior to offer 16A.

(Alternately, there is a mode in which it just eavesdrops on real TWCs instead of pretending to be one, but I don't use that mode and couldn't tell you much about what is/isn't supported running that way.)

ChutneyMary, 2022-09-23:
Thanks. That worked although I ended up using the Solar Surplus function with an enforced offset of -16 amps. The only reason for not using the Non-Scheduled behaviour is that the Amps Offered hunts around from 16 to 14. I'm guessing the spiking logic is still having some influence with this option.

To summarize, if anyone needs to disable the TWCManager automatic function -

1. Re-install TWCManager afresh to get rid of the Tesla API credentials,
2. Use the Schedule tab to enable Track Green Energy from 00:00,
3. Non-Scheduled power action to Track Green Energy,
4. Charge Rate Control - Use Tesla API Exclusively,
5. Consumption Offsets to -16A (or whatever rate you desire).

All the metrics I was previously monitoring via MQTT are still working fine for the alternative control system. 

I'll leave this topic open just as a prompt for the VIN "ignore list" (@ngardiner ). The only reason I'm using an alternative control system is to have discrete control over the two vehicles in the Home location.

MikeBishop, 2026-10-29:
Just as an update, what's in main now (with some improvements in #734) can now manage vehicles on both TWCs and UMCs properly. If you're able, give it a shit and let us know if you run into issues.

MikeBishop, 2026-10-30:
In 1.4.1 TWCManager treats a car on a Mobile Connector (or any other EVSE it doesn't control) as its own charging point: it gets its own share of the power and is only turned down or stopped on its own account, rather than being stopped along with the car on the Wall Connector. This ships in 1.4.1.

closed 2026-10-30 by MikeBishop