ngardiner/TWCManager #459: Enhancement: Allow for consumption offset based on vehicle SoC

alexeiw123, 2022-07-06:
Because I've got an export limit on my solar, I have my consumption offset so that my car is maximising the 'free' energy that would be export limited. This is great in summer and while it's sunny as there's plenty of surplus for the car.

During winter and rainy days, I'm better off charging the car from any solar, then I am exporting the solar then charging the car off peak.

It would be great if we could set a number of SoC (State of Charge) ranges that used different consumption offsets!

This way, I could have a nearly flat battery get prioritised charge and a nearly full battery just sip on the free electrons. Hopefully it's possible.

MikeBishop, 2022-07-06:
It's actually very possible -- you just have to do it in the config file instead of the UI.  Look at the [custom policy](https://github.com/ngardiner/TWCManager/blob/main/docs/PolicyCustomization.md) page.

So for example, I have a policy that says if the car has gotten very low, charge the battery to almost full:

        {
          "name": "Battery Low",
          "match": [
            "modules.TeslaPowerwall2.gridStatus",
            "modules.TeslaAPI.minBatteryLevelAtHome"
          ],
          "condition": [
            "eq",
            "lt"
          ],
          "value": [
            true,
            15
          ],
          "background_task": "checkCharge",
          "charge_amps": "config.wiringMaxAmpsAllTWCs",
          "charge_limit": 95,
          "latch_period": 300
        },

That says that if the battery level drops below 15%, set the charge limit to 95% and provide full power for five hours.

I also have two different green power policies -- the default one, which has a charge limit of 60%, and a second which triggers when the Powerwall is full with a charge limit of 90% for the car.

MikeBishop, 2022-07-06:
Actually, let me take that back -- it's possible to define multiple policies, but you can't currently specify a consumption offset in a policy.  That doesn't seem like a difficult feature request, though.

alexeiw123, 2022-07-09:
That sounds promising. Changing priority of solar from export to EV charging based on SoC would allow me to be pretty hands off.

I thought about using MQTT from home assistant and even using solcast forecast to do prediction but then realised I'd be way out of my depth 😂