ngardiner/TWCManager #460: Geolocation / „at home“ not working

danielsan1, 2022-07-06:
Hi,

since I switched to charge rate by API only and going back to work after corona ;) I have a problem charging at work 35km away from my home (charger). The car/TWCM via API is still controlling the charge rate here :-/ What can I do? I already changed the config setting homeradius to 1000 feet but that didn’t fix the problem.

Thanks in advance!

danielsan1, 2022-07-27:
Hey @ngardiner and/or @MikeBishop  - any idea on this? Would appreciate it :) 
(Sorry for approaching you directly)

ngardiner, 2022-07-27:
No need to apologise @danielsan1. I agree there appears to be an issue with our handling of home location as I have seen the same behaviour sporadically of late. I think we are corrupting the settings hash in some way, losing the true home location (and learning it again).

I am thinking of how we can best identify this one. I'll update soon.

danielsan1, 2022-07-27:
Thanks for your quick response! Glad that it’s not only an individual issue ;) Keep my fingers crossed

alexeiw123, 2022-09-04:
Could be related to this but I've noticed twice recently that my car has stopped charging at public fast chargers because the charge % has been set to 50%. I assume this TWCManager controlling while the car is not at home.

MikeBishop, 2026-10-26:
v1.4.0 includes a fix for home location being seeded or reset from a stale GPS fix (e9b50b7b, #423), which looks like the likely cause here. It also no longer wakes cars known to be away (#466). Could you re-verify on v1.4.0 whether at-home detection works now?

MikeBishop, 2026-10-27:
This should be resolved in 1.4.1. Where a car is now comes from plug-ins, unplugs, positions and drive events, and TWCManager applies a car's charge rate, limit or start only when it is plugged in at home or confirmed home. `atHomeRadius` defaults to 1,000 ft. If you still see TWCManager controlling a car charging away from home on 1.4.1, please reopen with a log.

closed 2026-10-27 by MikeBishop