ngardiner/TWCManager #462: TWC entity disappears from home assistant

alexeiw123, 2022-07-19:
I have TWC publishing to home assistant using a HA generated token.
For the most part this works well however I have a chart that plots my charger power. Occasionally (after a HA reboot) this entity vanishes for some period of time, which results in the chart not working anymore.

It seems like it comes back and remains once a car is plugged in again.

this is the entity I'm missing, there are other twc entities that remain: sensor.twcmanager_####_charger_load_w

[image]

bikeymouse, 2022-08-23:
Yeah, that is a result of the way the sensors are updated by TWC. That is done with a HTTP-post.
The result is that after a reboot of HA, the sensors are unavailable until they are updated.

I switched over to a REST sensor and MQTT sensors for this reason.

alexeiw123, 2022-08-23:
> Yeah, that is a result of the way the sensors are updated by TWC. That is done with a HTTP-post. The result is that after a reboot of HA, the sensors are unavailable until they are updated.
> 
> I switched over to a REST sensor and MQTT sensors for this reason.

Thanks, I've worked around this in the meantime with a simple template sensor in HA.

```
  - name: TWC Charger Load Always W
    state_class: measurement
    unit_of_measurement: W
    device_class: power
    state: >
      {% set value = states('sensor.twcmanager_2679_charger_load_w') %}
      {{ iif(is_number(value), value, 0) }}
```

ngardiner, 2022-08-26:
Thanks for the report - one thing we could do is to publish zero-values for all sensors per TWC at startup, this would fix the issue of the entity going missing before a vehicle charges, but doesn't fix the issue that if TWCManager isn't running the entities would not exist.

Sadly there's no HASS-side persistence option (and according to the following thread it won't be implemented), so without either an explicit sensor configured or using MQTT, it seems there's no way to do this with HTTP sensors: https://community.home-assistant.io/t/creating-permanent-sensors-to-post-to-using-http-api/378901/7

alexeiw123, 2022-08-28:
> Thanks for the report - one thing we could do is to publish zero-values for all sensors per TWC at startup, this would fix the issue of the entity going missing before a vehicle charges, but doesn't fix the issue that if TWCManager isn't running the entities would not exist, and it would also leave a gap where if you restart HASS after TWCManager was started these sensors wouldn't exist until the vehicle charges.

This sounds like a sound improvement in any case. Could TWCManager publish zero values by MQTT periodically to at least make errors not be long lasting?

In any case, my template sensor above is replacing missing entries with zero anyway so I don't see the issue anymore.

ngardiner, 2022-08-28:
We could certainly do that, the other option would be to periodically poll hass to confirm the existence of entities and publish null values if they don't exist.

The benefit being that we don't continue to spam hass with zero values for entities that exist but I guess that is vs the spam that is constantly querying for them, so it is a bit of a case of pick your poison.

My reading of the hass developers opinion on this is that this should be configured in hass config as you have done, so I believe your approach is the best practice as far as homeassistant is concerned.

closed 2026-04-26 by ngardiner