ngardiner/TWCManager #464: Don't charge Playtime because vehicle.stopAskingToStartCharging == True

Philip112112, 2022-08-10:
Hello,

I've installed the twcmanager without any issues, managed to save the bearer and refresh token, but can't get the damn thing to start charging. The only changes I've made to the config file, are the max Amps set to 16A. Below is a piece of the logs I get.
Any idea what I'm missing?

twcmanager_1  | 21:46:51 ⛽ Manager  12 Rx@: (F8) FD E0 29 43 77 77 08 06 40 00 00 00 00 00 00 88
twcmanager_1  | 21:46:51 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowAmps [16]), condition (gt), value (0)
twcmanager_1  | 21:46:51 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1660181656.4157908]), condition (gt), value (0)
twcmanager_1  | 21:46:51 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1660181656.4157908]), condition (gt), value (1660160811.3526223)
twcmanager_1  | 21:46:51 ⛽ Policy   14 All policy conditions have matched. Policy chosen is Charge Now
twcmanager_1  | 21:46:51 ⛽ Policy   10 Charge at 16.00
twcmanager_1  | 21:46:51 ⛽ Manager  20 Limiting charging to 16.00A.
twcmanager_1  | 21:46:51 ⛽ Manager  20 Charge when above 12A (minAmpsPerTWC).
twcmanager_1  | 21:46:51 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
twcmanager_1  | 21:46:51 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 21:46:51 ⛽ Slave    10 TWCID=29 43 desiredAmpsOffered=16 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=16 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=501
twcmanager_1  | 21:46:51 ⛽ Slave    10 set_last_amps_offered(TWCID=29 43, desiredAmpsOffered=16)
twcmanager_1  | 21:46:51 🚗 TeslaAPI 13 car_api_available returning True
twcmanager_1  | 21:46:51 ⛽ Manager  12 Rx@: (F8) FD E0 29 43 77 77 08 06 40 00 00 00 00 00 00 88
twcmanager_1  | 21:46:51 ⛽ Master   10 Number of cars charging now: 0
twcmanager_1  | 21:46:51 ⛽ Slave    10 TWCID=29 43 desiredAmpsOffered=16 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=16 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=501
twcmanager_1  | 21:46:51 ⛽ Slave    10 set_last_amps_offered(TWCID=29 43, desiredAmpsOffered=16)
twcmanager_1  | 21:46:52 🔌 RS485    12 Tx@: C0 FB E0 77 77 29 43 00 00 00 00 00 00 00 00 00 3A C0
twcmanager_1  | 21:46:52 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
twcmanager_1  | 21:46:52 🚗 TeslaAPI 13 car_api_available returning True
twcmanager_1  | 21:46:52 🚗 TeslaAPI 13 startOrStop is set to start
twcmanager_1  | 21:46:52 🚗 TeslaAPI 13 Don't charge Playtime because vehicle.stopAskingToStartCharging == True

ngardiner, 2022-08-11:
TWCManager is sending the start charging command to the vehicle but it doesn't appear to be starting. There's clearly no API access issue as it can see the vehicle fine.

This usually happens to me after a software upgrade (not every time, but often enough, less regularly of late) and requires me to do the 2 button + brake reboot method to get the car to respond to charge commands again. One test to validate if this is the issue or not is to try the start charging button on your phone app and see if it will get the car to charge or not.

Philip112112, 2022-08-12:
I've tried to start the charging with the app, that didn't work. I did manage to get it working by unplugging and plugging the cable back in the car. So for me, the issue is resolved :)

MikeBishop, 2022-11-05:
Closing as resolution doesn't indicate an issue with TWCManager.

closed 2022-11-05 by MikeBishop