ngardiner/TWCManager #465: How to use Tesla's scheduled charging with full amps (16)?

danielsan1, 2022-08-19:
After probably years I had to use the "scheduled charing" function of my Model 3. 
I was wondering how to use it... At first I tried "Charge now" but it overwrites the schedule function and immediately startet charging. Than I used the "Do not charge" setting. I think that's the right way but it stays at 6A (3-phase / Germany). Even when I set "minAmpsPerTWC" to 16A (it was 6A before). What am I doing wrong? :)
For now I can use it but to have to calculate the time for super slow 6A charging... 
Thanks!

KirkKirk, 2022-08-21:
It is well described in the manual, but look at my settings. It works flawlessly
 
[image]

Select which day to charge and that's it.
[image]

danielsan1, 2022-08-25:
Thanks! I’ll try this next time. 
I used the schedule for green energy charging during the day… but your explanation makes total sense :) I was always a bit confused with the schedule settings in general.

closed 2022-08-25 by danielsan1