ngardiner/TWCManager #468: MQTT status interface connects/disconnects 7 times per minute

bikeymouse, 2022-08-29:
Hi,
I'm using the MQTT status interface, which seems to work fine. I have set a rate limit to an update every 30 second, and indeed I see updates on Mosquito every 30 seconds.
However when I check the Mosquito log, I see that TWCManager is connecting/disconnecting about 7 times minute, effectively spamming Mosquito. 

All my other systems that publish on Mosquito only connect once, even if they publish multiple messages per minute.

Any way to prevent this?

I'm using TWCManager on Docker with twcmanager/twcmanager:latest tag, "V 1.3" which is the image from 22/04/2022

I'm running Mosquito in another container with a shared network and have disabled the password on Mosquitto so every subscriber can connect without authentication.

This is what I see every minute in the Mosquito log: 

[image]

ngardiner, 2022-08-29:
Hmmm, definitely doesn't look right to me. Is there any indication in the TWCManager logs as to why the connection is being terminated?

bikeymouse, 2022-08-30:
There are no errors in the TWC-log. And as said, it does successfully publish the messages. 
It’s almost like it sets up a new connection for every measurement it wants to publish. If that would be the case and it would use the same Id (which it does looking at the logs) then Mosquitto would disconnect the connection before accepting the next message, because there can’t be 2 producers/consumers with the same Id.
So it seems like a bug to me. 
Can you check if this is also not happening with your version?

bikeymouse, 2022-09-03:
I’m wondering if other users could check to see if this is common or specific to my set-up?
Anyone care to check the Mosquitto logs?

tjikkun, 2022-09-03:
https://github.com/ngardiner/TWCManager/blob/main/lib/TWCManager/Status/MQTTStatus.py#L138
So this is currently by design. It might be a good idea to change, but right now it works as "intended" it seems

bikeymouse, 2022-09-03:
Good find! 
Yes, this seems on purpose, but not really a good practice as this really burdens the broker (and TWC itself of course) with all the connection set-up/disconnects.
Never have seen a client behaving this way, so hopefully someone can change this behavior.

RichieB2B, 2023-06-02:
I just checked and TWC is using the same connection for all its status messages. It looks like this is fixed.