ngardiner/TWCManager #473: Unstable RS485 connection?

reklame33, 2022-09-18:
My charging is interrupted several times and I suspect this to be a unstable RS485 connection.
3 pcs TWCgen2 chargers are connected together to a RS485-to-ethernet device, (WuT COMserver).
In the log it shows sometimes that the 26 seconds limit exceeds and other messages with unexpected length or checksum error.

There are some parameters to tune in the WuT COMserver, but would like to ask here if any other suggestions is made.

log:

09:25:00 ⛽ Manager  12 Rx@: (FC) FD E2 40 85 1D 0C 80 00 00 00 00 00 00 00 00 50
09:25:00 ⛽ Manager  20 32.00 amp slave TWC 4085 is ready to link.  Sign: 1D
09:25:00 🔌 RS485    12 Tx@: C0 FB E0 77 77 40 85 09 0C 80 00 00 00 00 00 00 28 C0
09:25:00 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
09:25:00 🚗 TeslaAPI 13 car_api_available returning False because of recent carApiLasterrorTime 1663493100.1010776 or empty carApiBearerToken ''
09:25:00 🚗 TeslaAPI 13 car_api_charge return because car_api_available() == False
09:25:00 ⛽ Manager  12 Rx@: (FC) FD E2 24 31 2D 0C 80 00 00 00 00 00 00 00 00 F0
09:25:00 ⛽ Manager  20 32.00 amp slave TWC 2431 is ready to link.  Sign: 2D
09:25:00 🔌 RS485    12 Tx@: C0 FB E0 77 77 24 31 00 00 00 00 00 00 00 00 00 23 C0
09:25:00 ⛽ Manager  12 Rx@: (FC) FD E2 00 F2 8B 3F 47 46 F6 FF FF FF FF FF FF
09:25:00 ⛽ Manager  20 ERROR: Ignoring message of unexpected length 15: FD E2 00 F2 8B 3F 47 46 F6 FF FF FF FF FF FF
09:25:00 ⛽ Manager  10 Found end of message before full-length message received.  Discard and wait for new message.
09:25:00 ⛽ Manager  10 Found end of message before full-length message received.  Discard and wait for new message.
09:25:00 ⛽ Manager  10 Found end of message before full-length message received.  Discard and wait for new message.
09:25:01 ⛽ Manager  10 Found end of message before full-length message received.  Discard and wait for new message.
09:25:01 ⛽ Manager  12 Rx@: (FD E2 24 31 2D 0C 80 00 00 00 00 00 00 00 00 F0) FD E2 24 31 2D 0C 80 00 00 00 00 00 00 00 00 F0
09:25:01 ⛽ Manager  20 32.00 amp slave TWC 2431 is ready to link.  Sign: 2D
09:25:01 🔌 RS485    12 Tx@: C0 FB E0 77 77 24 31 00 00 00 00 00 00 00 00 00 23 C0
09:25:01 ⛽ Manager  12 Rx@: (FC) FD E2 00 32 6F 0C 80 00 00 00 00 00 00 00 00 0F
09:25:01 ⛽ Manager  20 32.00 amp slave TWC 0032 is ready to link.  Sign: 6F
09:25:01 ⛽ Manager  20 Set slave TWC 0032 protocolVersion to 2, minAmpsTWCSupports to 6.
09:25:01 🔌 RS485    12 Tx@: C0 FB E0 77 77 00 32 00 00 00 00 00 00 00 00 00 00 C0

09:31:59 🚗 TeslaAPI 13 car_api_charge return because car_api_available() == False
09:31:59 ⛽ Manager  12 Rx@: (FC) FD E2 80 7B 4C 13 B7 9F EF E7 FF FF FF FF
09:31:59 ⛽ Manager  20 ERROR: Checksum 865 does not match FF.  Ignoring message: FD E2 80 7B 4C 13 B7 9F EF E7 FF FF FF FF
09:32:01 ⛽ Manager  12 Msg timeout (FD E0 40 85 77 77 00 00 00 00 00 00 00 00 00 93) C0 FC
09:32:01 🔌 RS485    12 Tx@: C0 FB E0 77 77 00 32 00 00 00 00 00 00 00 00 00 00 C0
09:32:01 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API

dehsgr, 2022-09-18:
Check if this could be a duplicate of #461.

reklame33, 2022-09-18:
> Check if this could be a duplicate of #461.

Thanks, I saw this thread earlier and have implemented the parameter but get these errors in log anyway.

reklame33, 2022-09-18:
Here is a log when car stopped charging just now:

10:07:28 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowAmps [32]), condition (gt), value (0)
10:07:28 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1663496264.9880564]), condition (gt), value (0)
10:07:28 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowTimeEnd [1663496264.9880564]), condition (gt), value (1663495648.793211)
10:07:28 ⛽ Policy   14 All policy conditions have matched. Policy chosen is Charge Now
10:07:28 ⛽ Policy   10 Charge at 32.00
10:07:28 ⛽ Manager  20 Limiting charging to 32.00A.
10:07:28 ⛽ Manager  20 Charge when above 5A (minAmpsPerTWC).
10:07:28 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
10:07:28 🚗 TeslaAPI 13 car_api_available returning False because of recent carApiLasterrorTime 1663495648.7954261 or empty carApiBearerToken ''
10:07:28 🚗 TeslaAPI 13 applyChargeLimit return because car_api_available() == False
10:07:28 ⛽ Master   10 Number of cars charging now: 1
10:07:28 ⛽ Slave    10 desiredAmpsOffered TWC: 40 85 reduced from 32 to 15.309999999999999 with 1 cars charging and flex Amps of 0.0.
10:07:28 ⛽ Slave    10 TWCID=40 85 desiredAmpsOffered=15 spikeAmpsToCancel6ALimit=16 self.lastAmpsOffered=15 self.reportedAmpsActual=0.0 now - self.timeReportedAmpsActualChangedSignificantly=2570
10:07:28 ⛽ Slave    10 set_last_amps_offered(TWCID=40 85, desiredAmpsOffered=16)
10:07:28 ⛽ Master   10 Total amps all slaves are using: 16.69
10:07:28 ⛽ Master   10 Total amps all slaves are using: 16.69
10:07:28 ⛽ Slave    20 WARNING: Offering slave TWC 4085 15.0A instead of 16.0A to avoid overloading wiring shared by all TWCs.
10:07:28 ⛽ Slave    20 SHB 4085: 00 00.00/00.00A 0000 0000  M: 09 00.00/15.00A 0000 0000
10:07:28 ⛽ Slave    20 Slave power for TWCID 4085, status: 0
10:07:29 ⛽ Manager  20 WARNING: We haven't heard from slave 0032 for over 26 seconds.  Stop sending them heartbeat messages.
10:07:29 🔌 RS485    12 Tx@: C0 FB E0 77 77 40 85 09 05 DC 00 00 00 00 00 00 7D C0
10:07:29 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
10:07:29 🚗 TeslaAPI 13 car_api_available returning False because of recent carApiLasterrorTime 1663495649.7833111 or empty carApiBearerToken ''
10:07:29 🚗 TeslaAPI 13 car_api_charge return because car_api_available() == False
10:07:30 🔌 RS485    12 Tx@: C0 FB E0 77 77 24 31 09 05 DC 00 00 00 00 00 00 0D C0

closed 2026-04-25 by ngardiner