ngardiner/TWCManager #479: Two instances of TWCManager required in household

ChutneyMary, 2022-10-08:
I've installed a second Tesla Wall Charger including a second TWCManager on my home network. I can't launch the webpage and the response from the browser is a blank white page. Logging shows the following error -

19:52:04 🎮 HTTP     40 Unable to start HTTP Server: [Errno 98] Address already in use

I guessing that two web-servers on port 8080 in the same network could be a problem? If that would be an issue, where would I find the file to change one TWCManager to a different port?

Any other ideas on what could be wrong?

MikeBishop, 2022-10-09:
This shouldn't be a problem if the two instances of TWCManager are running on separate devices.  If they're on the same device, is there a reason not to simply have two TWCs managed by the same instance of TWCManager?

ChutneyMary, 2022-10-09:
I've got two instances of TWCManager running on two separate Raspberry Pi's (one in a front carport and one in a rear garage). I just can't work out why the web-page won't display. I've tried a complete fresh install a couple of times with no success. The log error 'Address already in use' is the only thing I've got to work with. A reboot of the home router has been done and I've double-checked there's not a duplication of the IP address on the two devices.
I've gone through all the files in the folder 'TWCManager' looking for instances of 8080 (if that is indeed the problem). I can't find where that port is defined in the code.

ChutneyMary, 2022-10-09:
The following is the error displayed when trying to run TWCManager. This is a newly imaged Raspberry Pi Zero W running "Raspberry Pi OS Lite (32-Bit)" - Debian Bullseye with no desktop environment and following the install instructions to the letter.

I've turned off the other Raspberry Pi as an interim measure and the "Unable to start HTTP Server" error has gone. Notwithstanding, the browser stills just display a white empty page. 

14:41:57 🎮 HTTP     20 Serving at port: 8080
14:41:57 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
14:41:58 ⛽ Manager  20 32.00 amp slave TWC 8783 is ready to link.  Sign: 60
14:41:58 ⛽ Manager  20 Set slave TWC 8783 protocolVersion to 2, minAmpsTWCSupports to 6.
14:41:58 ⛽ Policy   20 New policy selected; changing to Non Scheduled Charging
14:41:58 ⛽ Manager  20 Limiting charging to 0.00A.
14:41:58 ⛽ Manager  20 Charge when above 12A (minAmpsPerTWC).
14:41:58 ⛽ Slave    20 SHB 8783: 00 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
14:41:58 ⛽ Slave    20 Slave power for TWCID 8783, status: 0
14:41:58 ⛽ Manager  20 Slave TWC 8783: Delivered 57 kWh, voltage per phase: (251, 247, 248).
14:41:59 ⛽ Manager  20 32.00 amp slave TWC 8783 is ready to link.  Sign: 60
14:41:59 ⛽ Slave    20 SHB 8783: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
14:41:59 ⛽ Slave    20 Slave power for TWCID 8783, status: 9
14:42:01 ⛽ Slave    20 SHB 8783: 00 00.00/00.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
14:42:01 ⛽ Slave    20 Slave power for TWCID 8783, status: 0
----------------------------------------
Exception occurred during processing of request from ('10.1.1.8', 54529)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 160, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 781, in do_GET
    page = self.template.render(vars(self))
  File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/pi/TWCManager/lib/TWCManager/Control/themes/Modern/main.html.j2", line 1, in top-level template code
    {% extends "master.html.j2" %}
  File "/home/pi/TWCManager/lib/TWCManager/Control/themes/Modern/master.html.j2", line 11, in top-level template code
    {% block content %}{% endblock %}
  File "/home/pi/TWCManager/lib/TWCManager/Control/themes/Modern/main.html.j2", line 306, in block 'content'
    <label for="chargeNowRate" class="form-label text-muted">Charge with <span class="text-dark font-weight-bold" id="{{ twcid }}_ampsVal">{{ ampsMin[0] }}</span> A</label>
  File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 466, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: list object has no element 1

ChutneyMary, 2022-11-16:
I'm fairly sure something has changed in the 'Main' branch that is preventing the webpage from launching.

I've started with a clean installation using Raspberry Pi Imager (tried both Raspberry OS - with Desktop and Raspberry Pi OS Lite - without desktop). Then carried out the install instructions as per -

https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md

git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout v1.2.4
sudo make install

**or**

git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout main
make install

TWCManager works for both Main and v1.2.4 (ie. the terminal window shows the expected communications occuring). However the webpage fails to display if the Main branch is installed. I'd welcome anyone doing a clean install and following the install directions exactly as written. I'm using a Raspberry Pi Zero W.