ngardiner/TWCManager #480: Charging priority when 2  cars are charging.

KirkKirk, 2022-10-30:
I love TWCManager and am very grateful for your work @ngardiner and the other helpers.
I just got an idea for another In the scenario when 2 cars are connected at night to start charging when sufficient solar (10A or above), I noticed that both cars trying to charge but they are "fighting" and they going on and off until there is 18A or above.
Can we have the option to choose which car to start first, and when is like 20A solar excess second car to start? And when the solar exes drop below 18A, the second car to stop charging.
I think In this case "car fighting" will stop.

MikeBishop, 2022-11-08:
This is a part of the code I've looked at a fair bit.  What happens is that when both cars are trying to charge, they each try to take their "fair share" of the available power.  If half the available power is below the threshold to charge, TWCManager chooses to offer zero.

The trouble is that as soon as one car stops, it no longer counts toward the "fair share" calculation, and the remaining car (actually, the remaining TWC) concludes that it can take all the power, meaning it can charge.  But charging broadcasts to all cars, telling the other car to start again, so....

It's currently very difficult to allocate power to one TWC and not others.  I'm going to have to touch this code as part of #483, and I'll try to nudge it toward:

- Having a state where a TWC wants to charge but can't currently
- Allocating power to individual TWCs and leaving others at 0.

I'll note that the far more likely outcome, though, is that both cars will start charging at their minimum (5-6A) and draw some grid power until there's enough solar to cover both.

KirkKirk, 2022-11-09:
Thanks for looking at this. It is a tricky challenge. 
Just to add my thoughts related to my case in setup 2x chargers share 1 phase, 32A MAX Current (when both active 2x 16A)
Solar surplus is usually ~8.5kWh

Car 1                                            Car 2 
VIN provided                               VIN NOT provided
Charge 40%                                 Charge 25%

Option 1: Offer Amps to the less-charged car (Car2)
Option 2: Offer Amps to the more charged car (Car1)
Option 3: Offer Amps to both (not a good option as charging is less efficient)
if-then: When the charging car reaches the limit, offer Amps to both.

MikeBishop, 2022-11-16:
@KirkKirk, why doesn't Car2 report the VIN?  If it's not a Tesla, we presumably have no easy way to fetch the current charge level.  If it's an older TWC that can't report VINs, how do we know that's the car connected to that TWC versus an unknown vehicle?

It's actually a little more pernicious than that, even.  Unless charging is in progress, the best we know is the last-connected VIN, and that only if TWCManager hasn't restarted lately.  Making some guesses about which vehicle is "most in need" might be possible, but that's always going to be a low-fidelity decision.

MikeBishop, 2026-10-29:
Old bug report, but this should be improved after #734 merges. If there's only enough power for one vehicle, the other will stop until there's enough for it again.

MikeBishop, 2026-10-30:
In 1.4.1 power is shared among the charging stations whose minimum fits, with cars already charging kept first. If there's only enough for one car, the other waits until there's enough for it instead of the two taking turns starting and stopping. This ships in 1.4.1.

closed 2026-10-30 by MikeBishop

ngardiner, 2026-10-31:
Fixed in main by #734, shipping in 1.4.1: power goes first to cars already charging, and a second car starts only once there is enough for its minimum as well, so the two no longer take turns starting and stopping.