ngardiner/TWCManager #484: MQTT Connected repeatedly?

MikeBishop, 2022-11-08:
I'm seeing the "MQTT Connected" event firing over and over again; the Mosquitto logs say that it's recognizing the connection as a duplicate and closing the old connection, but TWCManager (or perhaps paho) seems to be interpreting that old connection closure as closing the new connection.

I'm not convinced this is a bug in TWCManager, but I'm having trouble pursuing it further right now:
[image]

[image]

tjikkun, 2022-11-08:
Related to #468

MikeBishop, 2022-11-08:
In my case, it turned out I'd mistakenly run a second instance of TWCManager which oddly didn't die from not being able to claim the HTTP port.  Since TWCManager uses a fixed client connection string, each one caused the other one to disconnect, which would reconnect and disconnect the first, and so on.

MikeBishop, 2022-11-08:
You should be able to set a different port number in your config file at control \ HTTP \ listenPort.  But as I said before, the two instances on two separate devices shouldn't have anything to do with one another, and you might want to figure out what's listening on that port instead.  Consider `netstat -ltnp` for that.