ngardiner/TWCManager #488: Upgrading TWCManager

MiRoCZ, 2022-11-09:
Hello, what I do wrong ? Trying to simply upgrade TWCManager 1.2.5 to 1.2.6 by command `sudo pip install --upgrade twcmanager` , but even everything seems to be OK an upgrade successfully executed, I still see v1.2.5. (v1.2.6 Available) at browser even after page refresh and cookies delete. I´d like to upgrade to 1.2.6 as I´m getting Error encountered during Phase 2 (POST) of the Tesla Authentication process.

[image]

[image]

And `pip show TWCManager` is showing version 1.2.6
[image]

Any ideas ? Thank you.

ngardiner, 2022-11-09:
Hi @MiRoCZ, you need to restart the service after the upgrade - try:

```sudo service twcmanager restart```

And see if that fixes it

MiRoCZ, 2022-11-09:
Hi Nathan @ngardiner, my issue described above is happening when I run TWCManager as script by command `sudo -u twcmanager python -m TWCManager`

I just tried to run TWCManager as service and then I can see finally version 1.2.6 and API token can be stored. Thank you for your help.

I just still wonder, why there is still 1.2.5 when TWCManager run as script ?

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop