ngardiner/TWCManager #490: lifetimecounter

magic7music, 2022-11-26:
Hi Guys,

do you know where I can find the lifetimecounter with date and time in the system? I saw it as a list sometime ago but cann´t find it anymore. I´m using 2.4 docker.
I want to use this data to get some THG quotes.

Thanks for replying

MikeBishop, 2022-11-27:
`/etc/settings.json`, I believe.

magic7music, 2022-11-27:
Thanks Mike,

this si what I found
...
{"Description": "Built-in Group - Vehicles in this Group cannot charge on managed TWCs", "Built-in": 1, "Members": []}}, "Vehicles": {"5YJ3E7EBXKF336258": **

{"chargeSessions": 787, "startkWh": 0, "totalkWh": 302}},

** "history": [["2022-11-23T11:10:00+01:00", 108.99286689419795], ["2022-11-23T11:15:00+01:00", 194.27027027027006], ["2022-11-23T11:50:00+01:00", 1317.973904109593], ["2022-11-23T11:55:00+01:00", 1459.8803496503538], ["2022-11-23T12:00:00+01:00", 1445.4431506849387], ["2022-11-23T12:05:00+01:00", 1462.4934722222283], ["2022-11-23T12:10:00+01:00", 1469.4110204081703], ["2022-11-23T12:15:00+01:00", 1460.2359589041155], ["2022-11-23T12:20:00+01:00", 1459.040412371141], ["2022-11-23T12:25:00+01:00", 1456.8438698630196], ["2022-11-23T12:30:00+01:00", 1455.447042253529], ["2022-11-23T12:35:00+01:00", 1450.4397241379372], ["2022-11-23T12:40:00+01:00", 1463.5785714285776], ["2022-11-23T12:45:00+01:00", 1458.0812068965577], ["2022-11-23T12:50:00+01:00", 1452.5308333333405], ["2022-11-23T12:55:00+01:00", 1457.006198630144], ["2022-11-23T13:00:00+01:00", 1455.042959183677], ["2022-11-23T13:05:00+01:00", 1470.3774482758702], ["2022-11-23T13:10:00+01:00", 1461.145853658541], ["2022-11-23T13:15:00+01:00", 1463.5125342465835], ["2022-11-23T13:20:00+01:00", 1465.710689655181], ["2022-11-23T13:25:00+01:00", 1464.4042857142936], ["2022-11-23T13:30:00+01:00", 1465.5390909090993], ["2022-11-23T13:35:00+01:00", 1463.5219047619107], ["2022-11-23T13:40:00+01:00", 1461.6949315068546], ["2022-11-23T13:45:00+01:00", 1459.510208333337], ["2022-11-23T13:50:00+01:00", 741.4540540540539]], "SlaveTWCs": {"6160": {"supportsVINQuery": 1}}, "respondToSlaves": 1, "chargeNowAmps": 6,
...

Is there a log with time and date from sessions in the past?
This seems to be my last session.
It should content all sessions in 2022.

Thanks 
Koen

MikeBishop, 2022-11-28:
The "history" field keeps ~48 hours of very granular data -- wattage per 5-minute timeslice. The one you list before that is TWCManager's own record.  IIRC, the TWC itself reports a lifetime usage number periodically.

magic7music, 2022-11-29:
so I have to install an kWh counter between the grid and the TWC Gen2.

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop