ngardiner/TWCManager #493: TWCManager overrides scheduled charge time/departure time

alexeiw123, 2022-12-26:
I have been trying to set my car for an early morning charge to 100% for a long trip, so wanted to disable the green tracking for a short period.

I tried to do this by setting the car as you would with any charger: through departure time or scheduled charging.
Then to have TWCManager allow a set charging rate when the car start. Whether using a 'charge now' value for 24 h, or changing the non scheduled power action to 'charge at specified non-scheduled rate'.

Unexpectedly, after a short while of the screen saying 'charging will start at 4am' or similair, the car would start charging at max rate.

The only way I was able to set my charging as I would like to, was to remove my API credentials from the settings file, so that TWC Manager is unable to start the charge.

Can we have an easy way to disable TWCManager temporarily to allow use of these functions when needed?

MikeBishop, 2023-01-01:
Basically, what happens is that TWCManager will both tell the TWC to offer power and tell the car to charge.  When Scheduled Departure is configured, the car doesn't accept offered power, but will respond to a command to charge.  We should perhaps not send Start Charging commands when Scheduled Departure is active.  That's easy to detect with Teslamate and presumably possible to detect with the regular API.

alexeiw123, 2023-06-21:
Hi @MikeBishop is there a possibility of adding this, or an option to disable using API to start the charging?
TWCManager keeps starting my charge when I don't want it to. I think if I remove the tokens it would have other undesirable consequences!

If it helps for testing I (think) I'm on the dev branch now.