ngardiner/TWCManager #496: Create docker without ttyusb0

reklame33, 2023-01-23:
I am trying to create a docker container from twcmanager but intend to use it with socket rs485 communication.
In the docker-compose file I tried using /dev/null:/dev/null but after downloading container I get error again:

Any solution how to create container without having a /dev/ttyUSB0 device?

home-twcmanager-1  | Traceback (most recent call last):
home-twcmanager-1  |   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
home-twcmanager-1  |     "__main__", mod_spec)
home-twcmanager-1  |   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
home-twcmanager-1  |     exec(code, run_globals)
home-twcmanager-1  |   File "/usr/src/TWCManager/TWCManager.py", line 34, in <module>
home-twcmanager-1  |     import TWCManager.TWCManager
home-twcmanager-1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 616, in <module>
home-twcmanager-1  |     modinstance = modclassref(master)
home-twcmanager-1  |   File "/usr/src/TWCManager/lib/TWCManager/Interface/RS485.py", line 52, in __init__
home-twcmanager-1  |     self.ser = self.serial.serial_for_url(self.port, self.baud, timeout=0)
home-twcmanager-1  |   File "/usr/local/lib/python3.7/dist-packages/serial/__init__.py", line 90, in serial_for_url
home-twcmanager-1  |     instance.open()
home-twcmanager-1  |   File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 325, in open
home-twcmanager-1  |     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
home-twcmanager-1  | serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
home-twcmanager-1 exited with code 1

MikeBishop, 2023-01-23:
You'll need to tell the serial module not to load, and instead activate a different interface module.  That's further down in the config file, IIRC.  If you're not able to find it, I can look more closely at which thing needs to be disabled/enabled.

tjikkun, 2023-01-23:
You can configure tcp as a port as well, see https://github.com/ngardiner/TWCManager/blob/main/etc/twcmanager/config.json#L225

MikeBishop, 2023-01-23:
Right, you'd want to set RS485 to `"enabled": false` and TCP to `"enabled": true`

reklame33, 2023-01-23:
No matter what I specified in docker-compose.yml I was not able to start container.
Found some info how to edit files inside a stopped container. Used this command:

docker run --volumes-from 76c425085e38 -it busybox

Then edited /etc/twcmanager/config.json using vi
Replaced /dev/ttyUSB0 with this:
"port": "socket://192.168.10.97:8000"

Exited busybox and edited the docker-compose.yml to have /dev/null:/dev/null as device.

Now I was able to start container successfully!