ngardiner/TWCManager #498: Tesla API "hiccup" prevent Model 3 from sleeping // Update to 1.3?

danielsan1, 2023-02-08:
Hi all! Especially @ngardiner and @MikeBishop :-)
A couple of days ago I recognized that my Model 3 would no longer go into sleep mode. I tried multiple restarts and also disconnected my data logger "Tronity.io". (I wasn't thinking about TWCManager in the first place) Still no sleep. So I contacted Tesla and they said "it's a 3rd party app connection". This was the decisive clue to TWCManager. I removed my token and - sleep! 
Very strange as I NEVER experienced such an issue with TWCManager...
So now I want to bring everything back to normal and was wondering if I should try the 1.3 developer version as you (among many other things) updated something regarding the Tesla API.
**How can I safely upgrade from 1.2.6 to 1.3?** Or would you say that's not a good idea?

Thanks in advance!

MikeBishop, 2023-02-10:
Well, my updated code is still in an unmerged branch.  I primarily use TeslaMate for monitoring, but I haven noticed that TWCManager's connection to TeslaMate occasionally falls off, and when it returns to API polling it does keep the car awake.  I was thinking that was a bug in my branch that I haven't sorted out yet, but it's certainly possible the bug exists in 1.2.x as well.

danielsan1, 2023-02-11:
I don't think it's related to my monitoring tool Tronity.io ... they are not connected to each other and worked side-by-side for years now. Usually Tronity caused sleep issues from time to time. But not this time :) 
So you would not recommend to upgrade to TWCManager 1.3? (because wouldn't change much with regards to the Tesla API implementation?)

MikeBishop, 2023-02-15:
FWIW, I just pushed two changes I think will improve the situation in my branch:  Reinitializing the Teslamate connection every 24 hours, and being less aggressive about API calls when Teslamate isn't being used. However, I don't think that says anything about what's in other branches, since the aggressive code is something I added.

danielsan1, 2023-05-21:
It happened again! :-/ No sleep because of any kind of Tesla API hiccup… I overwrote the API Key in TWCManager and it was gone/car went into sleep. How/Where can I check what TWCManager is doing with the API? What kind of calls it’s making?

MikeBishop, 2026-10-26:
The Tesla API polling this describes has been rewritten for the Fleet API since 1.2.6, including not waking or polling cars that are away or asleep. Closing as obsolete; if you still see a car kept awake on v1.4.0, please open a new issue with logs.

closed 2026-10-26 by MikeBishop