ngardiner/TWCManager #504: TWC stop controlling start/stop charging and set the home location

KirkKirk, 2023-05-01:
Hello @ngardiner, 

I am experiencing some difficulty with TWC. It seems that since last week, I am unable to start or stop the charging process, both manually and through the green button. I have tried deleting the settings.json and adding the API, as well as configuring the home location for both cars through the settings/dropbox. However, I am still unable to initiate or stop the charging process. Would it be possible for you to take a look and provide a solution? Thank you.

dehsgr, 2023-05-01:
19:06:24 🚗 TeslaAPI 20 ERROR: Can't access vehicle status for MyTesla.  Will try again later.

…is written to my logs. I could imagine that this could be the cause. Maybe Tesla changed sth. at their site? 😳

dehsgr, 2023-05-01:
@ngardiner @MikeBishop I just enabled debug logging an this should be the root cause:
> 19:17:24 urllib3.co 10 https://owner-api.teslamotors.com:443 "GET /api/1/vehicles/ID/data_request/charge_state HTTP/1.1" 404 34
> 19:17:24 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/ID/data_request/charge_state <Response [404]>
> 19:17:24 🚗 TeslaAPI 10 Car API vehicle status{'status': 404, 'error': 'Not Found'}

Seems that there are really changes at API site. What can we do to get start/stop charging working again? 

Note: I'm running TWCManager in dockers testing branch on v1.3.0.

dehsgr, 2023-05-01:
@ngardiner @MikeBishop as of [https://tff-forum.de/t/tesla-api-charge-state-nicht-mehr-abrufbar/276390/2 Tesla seems to change their API at a lot of ](https://tff-forum.de/t/teslalogger-mit-raspberry-pi-mysql-grafana-osm-teil-2/209751/1348) Tesla deleted such specific endpoints and https://owner-api.teslamotors.com/api/1/vehicles/#VID/vehicle_data has to be used from now. Can you provide a quick fix for this?

dehsgr, 2023-05-01:
I‘ll check this out tomorrow. I’ve to patch my docker instance with new code manually, I assume.

Many thanks for the hotfix!

KirkKirk, 2023-05-02:
Hello @MikeBishop, I have received the file and manually replaced it. However, it seems that the issue still persists. Have you tested the system after deleting the setting file?

TitusJohannes, 2023-05-03:
Hello,
I updated my dockers installation with 
"sudo docker-compose -f docker-compose.yml down
sudo docker-compose -f docker-compose.yml pull
sudo docker-compose -f docker-compose.yml up -d"
But still, I get it run only if I set in the "onlyChargeMultiCarsAtHome": false". The "homeLat": XX.573428, "homeLon": XX.993959, is set correctly, is not changed, and was checked and verified via googlemaps.
Is there somebody with the same problem: I have a MS 2023.12.1.1 and MX 2023.6.11 with both cars, same problem?
Thanks for your help.
Yours, Titus

rutiruti, 2023-05-07:
get_room_mapping

rutiruti, 2023-05-07:
get_room_mapping

TitusJohannes, 2023-05-08:
Hello rutiruti, I am not sure to whom and waht exactly you want to say with the single command "get_room_mapping"?
...

KirkKirk, 2023-05-09:
I am still experiencing issues with TWC. It seems that the problem is linked to retrieving data from the account in both ways. Unfortunately, the location of the cars cannot be detected, meaning that they cannot be charged in Track Green Energy or Manual mode. It appears that the API has been successfully integrated, and both cars are visible in the dropbox. However, it is possible that the structure of the Tesla system has recently been altered and that should be checked.

MikeBishop, 2023-05-11:
I've merged the PR, since it seems to be working for @dehsgr (and me, though I'm on a different branch). Hopefully that will get it in the Docker image and updating the image will be sufficient?

dehsgr, 2023-05-11:
@TitusJohannes I suggest you copy TeslaAPI.py directly into your docker container and check again. See https://github.com/ngardiner/TWCManager/issues/504#issuecomment-1531388583

KirkKirk, 2023-05-13:
@MikeBishop , I am glad to inform you that everything has returned to its normal state after the recent update. Therefore, I will proceed to close this issue as it appears to be resolved.

THANKS.

closed 2023-05-13 by KirkKirk

TitusJohannes, 2023-05-29:
Dear Community,
I want to thank you so such much for fixing this issue. I do highly appreciate the efforts the community undergoes in order to keep this tool up to date, despite the fact, that Tesla terribly frequently changes Auth-Flows, change or discontinue commands.
If there is any donation box for the programmers of TWC Manager, I will definitely put something in…

Thank you once again,
Titus

PS at a first glance, I thought there is something wrong, because I installed the latest 1.31 version, and still the “1.30” version appears in the GUI. But everything works fine now and this is nothing really important….