ngardiner/TWCManager #510: 6A is too high for 3 phase charging

RichieB2B, 2023-05-15:
In [TWCManager.py](https://github.com/ngardiner/TWCManager/blob/main/lib/TWCManager/TWCManager.py#L1026) `minAmpsTWCSupports` is set to 6A (or 5A) for reasons explained in the [README](https://github.com/ngardiner/TWCManager/blob/main/docs/README.md#why-6a) and [TWCSlave.py](https://github.com/ngardiner/TWCManager/blob/main/lib/TWCManager/TWCSlave.py#L760). However, they only consider 1 phase charging. With 3 phase charging using a minimum of 6A means 4 kW which makes it very hard to effectively load balance a small PV setup.

Using a [different load balancer](http://github.com/RichieB2B/tesla-loadbalancer) I have successfully set the minimum to 1A using the the Tesla API. My Model S75D did not complain.

I think we should set `minAmpsTWCSupports` to 1A for 3 phase charging. After all 5A, 1 phase 120V is 600W and 1A, 3 phase 230 is 690W. Users still control `minAmpsPerTWC` in `config.json` so for 3 phase 120V they can set this to a safe value (the default is 12).

MikeBishop, 2023-05-15:
Yes, in principle, this seems reasonable.  What I'd want to see tested first is actually sending a command to go down to 1A, not via the API, but via the serial connection.  Neither of the maintainers of this project (absentee though we sometimes are) have a three-phase installation, so improvements for three-phase typically wind up coming from contributors who have the hardware to test a change.

To your change specifically, you're reading the config file to get the number of phases.  Those values are intended as fallbacks for the TWCs which can't tell us the reality directly, so you'd do better to see whether the TWC is reporting voltage and then read the config if not.

RichieB2B, 2023-05-15:
Good call, I updated the patch to use the actual voltages to determine a 3-phase TWC. All I need now is a sunny day to test it.

dehsgr, 2023-05-17:
@RichieB2B thanks for your work on this! I think your changes aren't enough. Reflecting https://github.com/RichieB2B/tesla-loadbalancer/blob/main/tesla-mqtt-loadbalancer.py:

```
    # set it twice if < 5A, see https://github.com/tdorssers/TeslaPy/pull/42
    if amps < 5:
      vehicle.command('CHARGING_AMPS', charging_amps=amps)
      time.sleep(5)
    result = vehicle.command('CHARGING_AMPS', charging_amps=amps)
```

...you might have to change API call for settings amps is being executed twice. See line

RichieB2B, 2023-05-17:
Yes, that will work for the API but I switched to TWCManager so I can use RS485 as it is much faster. It seems the TWC will not go lower than 6A even if told to so so via RS485.

dehsgr, 2023-05-17:
Yes, to reach that target I think we would have to use API in addition. 😕

commit 2120fdca (2023-05-27): Set minAmpsPerTWC = 1 for 3 phase chargers, fixes #510
closed 2023-05-27 by ngardiner