ngardiner/TWCManager #516: State of Charge not reporting correctly

alexeiw123, 2023-05-26:
have noticed the state of charge is not reporting correctly anymore. Given my installation environment hasn't changed, it seems more likely to be API related to me.

[image]

ngardiner, 2023-05-27:
My apologies for the delays around fixing the docker builds (and other things). I have to fix some tokens which have expired - I am working on it now and will update when it is resolved.

ngardiner, 2023-05-27:
Docker latest image is updated now

alexeiw123, 2023-05-27:
> Docker latest image is updated now

Awesome.

I'm definitely not doing this often enough to be fresh on what I'm doing.

As it turns out, I got my projects confused and I'm not running TWCManager in a docker environment.

So the next question is, what's the correct upgrade path these days?

I'm doing `sudo pip install --upgrade twcmanager`

and getting a lot of `Requirement already satisfied, skipping upgrade`

Do I need to change to development build to capture these API changes? If so, how?

alexeiw123, 2023-05-29:
> This is likely the same as #504, which broke basically all API-based information retrieval about the vehicle. The fix (#505) is merged, but the Docker builds are broken in a way I can't easily fix without @ngardiner's help. If you're able to manually pull `main`, you should be okay; if you're reliant on a Docker image, we'll have to wait for @ngardiner to assist.

Hi Mike,

Pulling main has upgraded my install but still seeing the same issue.

[image]

MikeBishop, 2026-10-26:
The original problem was fixed by #505 (switch to the `vehicle_data` endpoint, v1.3.0). @dehsgr, the surplus-charging problem you saw after upgrading to main looks unrelated; could you open a new issue with logs? Closing this one.

closed 2026-10-26 by MikeBishop