ngardiner/TWCManager #518: V1.3.0 - Export meter not working as expected with TWCManager

cods4, 2023-05-30:
I have recently got this running again after installing a second TWC. In the process I decided to re-install raspbian etc and the latest V1.3.0 of TWCManager and reconfigure TWCManager from scratch.

I am almost certain that before these changes, I was giving TWCM a single Consumption value from home assistant called sensor.main_power. 
This was coming directly from a Shelly EM3 which monitors my grid current, voltage, power etc.
When my house and solar system is exporting to the grid it reads negative power, and when I am buying energy from the grid it shows positive power.

When I configure this HASS entity as a Consumption meter under the HASS source, TWCM doesn't behave at all like I think it did before. When we are exporting power, everything reads zero, and when we are importing power I get positive consumption readings.

I have also tried the MQTT source using the some value taken directly from the Shelly EM3 and I got the same results.

What I want to do with my main_power sensor is have it feed directly into the 'solar surplus' variable in TWCM when it's negative, and into the Consumption variable when it is positive.
What am I doing wrong here?

Here's my config.json

``{
    "config": {
        # The settings path will store the dynamic settings which are defined via the web
        # interface
        "settingsPath": "/etc/twcmanager",

        # Set wiringMaxAmpsAllTWCs to the maximum number of amps your charger wiring
        # can handle. I default this to a low 6A which should be safe with the minimum
        # standard of wiring in the areas of the world that I'm aware of.
        # Most U.S. chargers will be wired to handle at least 40A and sometimes 80A,
        # whereas EU chargers will handle at most 32A (using 3 AC lines instead of 2 so
        # the total power they deliver is similar).
        # Setting wiringMaxAmpsAllTWCs too high will trip the circuit breaker on your
        # charger at best or START A FIRE if the circuit breaker malfunctions.
        # Keep in mind that circuit breakers are designed to handle only 80% of their
        # max power rating continuously, so if your charger has a 50A circuit breaker,
        # put 50 * 0.8 = 40 here.
        # 40 amp breaker * 0.8 = 32 here.
        # 30 amp breaker * 0.8 = 24 here.
        # 100 amp breaker * 0.8 = 80 here.
        # IF YOU'RE NOT SURE WHAT TO PUT HERE, ASK THE ELECTRICIAN WHO INSTALLED YOUR
        # CHARGER.
        "wiringMaxAmpsAllTWCs": 32,

        # If all your chargers share a single circuit breaker, set wiringMaxAmpsPerTWC
        # to the same value as wiringMaxAmpsAllTWCs.
        # Rarely, each TWC will be wired to its own circuit breaker. If you're
        # absolutely sure your chargers each have a separate breaker, put the value of
        # that breaker * 0.8 here, and put the sum of all breakers * 0.8 as the value of
        # wiringMaxAmpsAllTWCs.
        # For example, if you have two TWCs each with a 50A breaker, set
        # wiringMaxAmpsPerTWC = 50 * 0.8 = 40 and wiringMaxAmpsAllTWCs = 40 + 40 = 80.
        "wiringMaxAmpsPerTWC": 32,

        # https://teslamotorsclub.com/tmc/threads/model-s-gen2-charger-efficiency-testing.78740/#post-1844789
        # says you're using 10.85% more power (91.75/82.77=1.1085) charging at 5A vs 40A,
        # 2.48% more power at 10A vs 40A, and 1.9% more power at 20A vs 40A.  This is
        # using a car with 2nd generation onboard AC/DC converter (VINs ending in 20000
        # and higher).
        # https://teslamotorsclub.com/tmc/threads/higher-amp-charging-is-more-efficient.24972/
        # says that cars using a 1st generation charger may use up to 30% more power
        # at 6A vs 40A!  However, the data refers to 120V 12A charging vs 240V 40A
        # charging. 120V 12A is technically the same power as 240V 6A, but the car
        # batteries need 400V DC to charge and a lot more power is wasted converting
        # 120V AC to 400V DC than 240V AC to 400V DC.
        #
        # The main point is 6A charging wastes a lot of power, so we default to charging
        # at a minimum of 12A by setting minAmpsPerTWC to 12. I picked 12A instead of 10A
        # because there is a theory that multiples of 3A are most efficient, though I
        # couldn't find any data showing that had been tested.
        #
        # Most EU chargers are connected to 230V, single-phase power which means 12A is
        # about the same power as in US chargers. If you have three-phase power, you can
        # lower minAmpsPerTWC to 6 and still be charging with more power than 12A on
        # single-phase.  For example, 12A * 230V * 1 = 2760W for single-phase power, while
[... 648 lines cut ...]
            "ratelimit": 10,

       # Username and password are optional. If your broker requires a username and password to authenticate
       # clients, set them here. Otherwise, leave them commented out
            "username": "mqttuser",
            "password": "mqttpassword"
        }
    },
    "vehicle": {
        "TeslaMate": {
            "enabled": true,
            "syncTokens": false,
            "db_host": "192.168.1.1",
            "db_name": "teslamate",
            "db_pass": "teslamate",
            "db_user": "teslamate",
            "syncTelemetry": true,
            "mqtt_host": "192.168.21.30",
            "mqtt_user": "mqttuser",
            "mqtt_pass": "mqttpassword",
            "mqtt_prefix": "teslamate"
        }
    }
}
`

cods4, 2023-06-03:
Any ideas? Or am I simply using this wrong?

@ngardiner @MikeBishop

MikeBishop, 2023-06-05:
I vaguely recall a charge a while back to treat negative values as the opposite thing, which might have impacted your setup. But that seems to be exactly what you're doing here, so I'm not sure.  How do the logs say it's actually interpreting the values it's getting?

cods4, 2023-06-05:
How would I see how it is interpreting the value? I have logging level set to 9.
I can see it is getting the negative value, but that seems to be all.
The WebUI shows 0W for Generation and Consumption.

I am pretty sure when I was running an earlier version of TWCManager, the negative consumption values got interpreted as excess solar. Not sure if this is a configuration issue on my end, or something has changed in the code.

Logfile:

2023-06-06 10:55:20,925 - HASS       10 getConsumption returns -3035.44
2023-06-06 10:55:20,929 - HASS       10 Generation Entity Not Supplied. Not Querying
2023-06-06 10:55:20,935 - ⛽ Master   10 Number of cars charging now: 0
2023-06-06 10:55:20,939 - ⛽ Master   10 Total amps all slaves are using: 0.0
2023-06-06 10:55:20,945 - 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
2023-06-06 10:55:20,950 - 🚗 TeslaAPI 13 car_api_available returning True
2023-06-06 10:55:20,954 - 🚗 TeslaAPI 09 applyChargeLimit return because under 60 sec since last carApiLastChargeLimitApplyTime
2023-06-06 10:55:20,961 - ⛽ Master   10 Total amps all slaves are using: 0.0
2023-06-06 10:55:20,966 - ⛽ Manager  20 Limiting charging to 0.00A - 0.00A = 0.00A.
2023-06-06 10:55:20,971 - ⛽ Manager  20 Charge when above 8A (minAmpsPerTWC).
2023-06-06 10:55:20,977 - 📊 MQTT     10 MQTT Status: Attempting to Connect
2023-06-06 10:55:21,002 - 📊 MQTT     10 Connected to MQTT Broker with RC: 0
2023-06-06 10:55:21,007 - 📊 MQTT     09 Copy Message Buffer
2023-06-06 10:55:21,011 - 📊 MQTT     09 Clear Message Buffer
2023-06-06 10:55:21,015 - 📊 MQTT     13 Publishing MQTT Topic TWC/config/minAmpsPerTWC (value is 8)
2023-06-06 10:55:21,021 - 📊 MQTT     13 Publishing MQTT Topic TWC/all/maxAmpsForSlaves (value is 0)

cods4, 2023-06-25:
This is still an issue AFAIK. 
As a work around I have set up a template sensor in Home Assistant to convert negative power flow into green energy produciton, and a separate sensor for positive power/consumption.