ngardiner/TWCManager #521: Can't update to 1.3

danielsan1, 2023-06-04:
Hi! I was super excited to see that Version 1.3 is out now! But unfortunately I'm not able to upgrade :-/ I hope you can help me with that...

At first I used udo pip3 install --upgrade twcmanager as I didn't see the auto-update feature in the web interface. 
It ended with: 
> sentry-sdk 1.25.0 has requirement urllib3>=1.26.11; python_version >= "3.6", but you'll have urllib3 1.24.1 which is incompatible.
Installing collected packages: pyyaml, growattServer, sentry-sdk, typing-extensions, pkgutil-resolve-name, zipp, importlib-resources, pyrsistent, importlib-metadata, attrs, jsonschema, ocpp, six, pymodbus, solaredge-modbus, psycopg2, twcmanager
  Found existing installation: six 1.12.0
    Not uninstalling six at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'six'. No files were found to uninstall.
  Found existing installation: TWCManager 1.2.3
    Uninstalling TWCManager-1.2.3:
      Successfully uninstalled TWCManager-1.2.3
Successfully installed attrs-23.1.0 growattServer-1.4.0 importlib-metadata-6.6.0 importlib-resources-5.12.0 jsonschema-4.17.3 ocpp-0.19.0 pkgutil-resolve-name-1.3.10 psycopg2-2.9.6 pymodbus-2.5.3 pyrsistent-0.19.3 pyyaml-6.0 sentry-sdk-1.25.0 six-1.16.0 solaredge-modbus-0.7.0 twcmanager-1.3.0 typing-extensions-4.6.3 zipp-3.15.0

Then I saw the auto-update link in the top right corner of the web interface and clicked it. Unfortunately the output was:
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: TWCManager in /usr/local/lib/python3.7/dist-packages (1.3.0)
Requirement already satisfied, skipping upgrade: cryptography<3.4 in /usr/lib/python3/dist-packages (from TWCManager) (2.6.1)
Requirement already satisfied, skipping upgrade: jinja2>=2.11.2 in /usr/local/lib/python3.7/dist-packages/Jinja2-2.11.2-py3.7.egg (from TWCManager) (2.11.2)
Requirement already satisfied, skipping upgrade: solaredge-modbus>=0.7.0 in /usr/local/lib/python3.7/dist-packages (from TWCManager) (0.7.0)
Requirement already satisfied, skipping upgrade: pyserial>=3.4 in /usr/local/lib/python3.7/dist-packages/pyserial-3.5-py3.7.egg (from TWCManager) (3.5)
Requirement already satisfied, skipping upgrade: ocpp in /usr/local/lib/python3.7/dist-packages (from TWCManager) (0.19.0)
Requirement already satisfied, skipping upgrade: pyyaml in /home/pi/.local/lib/python3.7/site-packages (from TWCManager) (6.0)
Requirement already satisfied, skipping upgrade: requests>=2.23.0 in /usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg (from TWCManager) (2.25.1)
Requirement already satisfied, skipping upgrade: sysv-ipc in /usr/local/lib/python3.7/dist-packages/sysv_ipc-1.1.0-py3.7-linux-armv7l.egg (from TWCManager) (1.1.0)
Requirement already satisfied, skipping upgrade: termcolor>=1.1.0 in /usr/local/lib/python3.7/dist-packages/termcolor-1.1.0-py3.7.egg (from TWCManager) (1.1.0)
Requirement already satisfied, skipping upgrade: paho-mqtt>=1.5.0 in /usr/local/lib/python3.7/dist-packages/paho_mqtt-1.5.1-py3.7.egg (from TWCManager) (1.5.1)
Requirement already satisfied, skipping upgrade: pyModbusTCP>=0.1.8 in /usr/local/lib/python3.7/dist-packages/pyModbusTCP-0.1.10-py3.7.egg (from TWCManager) (0.1.10)
Requirement already satisfied, skipping upgrade: sentry-sdk>=0.11.2 in /usr/local/lib/python3.7/dist-packages (from TWCManager) (1.25.0)
Requirement already satisfied, skipping upgrade: growattServer>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from TWCManager) (1.4.0)
Requirement already satisfied, skipping upgrade: websockets>=9.1; python_version >= "3.7" in /usr/local/lib/python3.7/dist-packages/websockets-9.1-py3.7-linux-armv7l.egg (from TWCManager) (9.1)
Requirement already satisfied, skipping upgrade: psycopg2 in /usr/local/lib/python3.7/dist-packages (from TWCManager) (2.9.6)
Requirement already satisfied, skipping upgrade: pymysql in /usr/lib/python3/dist-packages (from TWCManager) (0.9.3)
Requirement already satisfied, skipping upgrade: MarkupSafe>=0.23 in /usr/local/lib/python3.7/dist-packages/MarkupSafe-2.0.0rc2-py3.7-linux-armv7l.egg (from jinja2>=2.11.2->TWCManager) (2.0.0rc2)
Requirement already satisfied, skipping upgrade: pymodbus>=2.4.0 in /usr/local/lib/python3.7/dist-packages (from solaredge-modbus>=0.7.0->TWCManager) (2.5.3)
Requirement already satisfied, skipping upgrade: jsonschema<5.0.0,>=4.4.0 in /usr/local/lib/python3.7/dist-packages (from ocpp->TWCManager) (4.17.3)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests>=2.23.0->TWCManager) (2018.8.24)
Requirement already satisfied, skipping upgrade: chardet<5,>=3.0.2 in /usr/lib/python3/dist-packages (from requests>=2.23.0->TWCManager) (3.0.4)
Requirement already satisfied, skipping upgrade: idna<3,>=2.5 in /usr/lib/python3/dist-packages (from requests>=2.23.0->TWCManager) (2.6)
Collecting urllib3<1.27,>=1.21.1 (from requests>=2.23.0->TWCManager)
  Downloading https://files.pythonhosted.org/packages/c5/05/c214b32d21c0b465506f95c4f28ccbcba15022e000b043b72b3df7728471/urllib3-1.26.16-py2.py3-none-any.whl (143kB)
Requirement already satisfied, skipping upgrade: six>=1.15.0 in /usr/local/lib/python3.7/dist-packages (from pymodbus>=2.4.0->solaredge-modbus>=0.7.0->TWCManager) (1.16.0)
Requirement already satisfied, skipping upgrade: pkgutil-resolve-name>=1.3.10; python_version < "3.9" in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (1.3.10)
Requirement already satisfied, skipping upgrade: typing-extensions; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (4.6.3)
Requirement already satisfied, skipping upgrade: attrs>=17.4.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (23.1.0)
Requirement already satisfied, skipping upgrade: importlib-metadata; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (6.6.0)
Requirement already satisfied, skipping upgrade: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (0.19.3)
Requirement already satisfied, skipping upgrade: importlib-resources>=1.4.0; python_version < "3.9" in /usr/local/lib/python3.7/dist-packages (from jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (5.12.0)
Requirement already satisfied, skipping upgrade: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata; python_version < "3.8"->jsonschema<5.0.0,>=4.4.0->ocpp->TWCManager) (3.15.0)
Installing collected packages: urllib3
Successfully installed urllib3-1.26.16

I manually went back to the index page and saw that I'm still on v 1.2.6... what can I do? Did I miss a step? 
(Regarding "requirement urllib3<1.27,>=1.21.1, but you'll have urllib3 2.0.2 which is incompatible." - 2.0.2 seems to be the latest version for my raspberry OS :( )

Thanks in advance!

ChutneyMary, 2023-06-07:
I'm also having problems installing the latest 1.3.0.

This is starting out on a completely fresh installation of Raspberry Pi OS Lite (Debian Bullseye with no desktop environment on a Raspberry Pi Zero W). The script, when running in the foreground, gives the following error message -

/usr/lib/python3/dist-packages/requests/__init__.py:87: RequestsDependencyWarning: urllib3 (2.0.3) or chardet (4.0.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "

The web-page launches successfully, but when attempting to save the Tesla API using the cut/paste of the URL, the web-page shows -
**This page isn't working
10.1.1.21 didn't send any data
(ERR_EMPTY_RESPONSE)**

I'd be very appreciative if anyone else can do a clean install using the instructions provided and see what's required to fix it.

ChutneyMary, 2023-06-07:
And some more info when observing the script when launching the web-page -

Exception occurred during processing of request from ('10.1.1.8', 58039)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 985, in do_POST
    res = master.getModuleByName("TeslaAPI").saveApiToken(url)
  File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1058, in saveApiToken
    params = json.loads(req.text)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

ngardiner, 2023-06-07:
Hi @ChutneyMary, this error occurs when the modules are installed via apt rather than pip - did you run setup.py? It should have upgraded those modules to the version specified by module dependencies.

ngardiner, 2023-06-07:
Hi @danielsan1, the issue is specific to the sentry_sdk module (which you are likely not using as it is an exception reporting platform primarily for development). I will update the dependencies to downgrade sentry_sdk for incompatible python versions, but you should not run into issues if you restart.

ChutneyMary, 2023-06-07:
I followed the instructions at (https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md)

sudo apt-get update
sudo apt-get install -y git python3 python3-pip python3-setuptools python3-dev libatlas-base-dev

then

git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout v1.2.4
sudo make install

(changing git checkout to v1.3.0)

ChutneyMary, 2023-06-08:
Apologies for being that numpty linux newb....I should clarify that I'm not seeing any success following the setup instructions at https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md

@ngardiner - You asked 'did you run setup.py?'. How do I run setup.py and where in the installation steps should that occur? (assuming a new install on Raspberry Pi OS Lite?)

danielsan1, 2023-06-08:
Hi @ngardiner thanks for clarifying! Strangely, after a service restart (and even a system reboot) its still version 1.2.6 ... 
When I check piip3 show - it's 1.3.0 

> pi@twcmanager:~ $ pip3 show TWCManager
Name: TWCManager
Version: 1.3.0
Summary: Package to manage Tesla Wall Connector installations
Home-page: https://github.com/ngardiner/twcmanager
Author: Nathan Gardiner
Author-email: ngardiner@gmail.com
License: None
Location: /usr/local/lib/python3.7/dist-packages
Requires: sysv-ipc, websockets, growattServer, requests, jinja2, ocpp, cryptography, termcolor, pyserial, pymysql, sentry-sdk, solaredge-modbus, paho-mqtt, pyyaml, psycopg2, pyModbusTCP
Required-by: 

Very strange... what can I do? Any ideas?

As always - thanks for your great support! :-)

ngardiner, 2023-06-11:
@ChutneyMary I managed to replicate the issue in bullseye, and I'll push an update shortly.

You should be able to fix it by editing requirements.txt and changing:

```requests>=2.23.0```
to
```requests>=2.25.2```

And then running ```make install``` again. This overrides the version installed by debian package (python3-requests).

I'll test against the latest Debian version (bookworm) and if it's working there as well, I'll push it out.

danielsan1, 2023-06-13:
@ngardiner do you have any additional idea/suggestion for my strange issue? (Different way to update/install maybe?)

danielsan1, 2023-07-07:
Hey @ngardiner @MikeBishop do you have any additional idea/suggestion for my strange issue? (Different way to update/install maybe?)