ngardiner/TWCManager #522: NodeRed fetching Consumption and Generation data via URL

tomd135, 2023-06-07:
I am currently struggling to fetch consumption and generation data via URL.
My Victron system with NodeRed is providing the data with a response node.

Manually tested in a Web Browser I get the following result.
https://192.168.123.102:1881/solar/consumption
Result: 234.6
https://192.168.123.102:1881/solar/generation
Result: 4801.000071540475

Here is my config /etc/twcmanager/config.json
` "URL": {
          "enabled": true,
          "url": "https://192.168.123.102:1881/solar",
          "consumptionItem": "consumption",
          "generationItem": "generation"
        } 
`

Looks good from my point of view, but unfortunately TWCmanager is reporting nothing.
What is going wrong? Thanks for your help.

dehsgr, 2023-06-07:
You're right. This looks good. BUT: You're using HTTPS. Is the certificate from https://192.168.123.102 tursted to TWCManager system? This might be the issue. You could (temporarily) try to use HTTP for connection to verify this.

tomd135, 2023-06-07:
Hi dehsgr,
I am using NodeRed out of my Victron System and I believe I have no chance to change it to HTTP. Tried it out but it is not accepted.
But alternatively tried to deal with the certificate:
-copied the viltron certificate from the viltron NodeRed Raspi to the TWCmanager Raspi
  from this path: `/usr/local/share/ca-certificates/victronenergy`
  into the same path on the other machine
-install certificate with `sudo update-ca-certificates`
it answers with 1 added certificate

Unfortunately TWCmanager still does not show the consumption and generation values yet.

Was that the wrong way or incomplete? 
Thanks Thomas

dehsgr, 2023-06-07:
Hmmm. Okay. Do you have solar surplus profile active? This must be active for showing data.

tomd135, 2023-06-08:
Yes, tried also to get generation data from my Fronius. That works, but consumption is still missing.
Any idea what I could try?
Thanks Thomas

dehsgr, 2023-06-08:
Do you have TWCManager installation running directly or within docker container? You could try to query URLs via curl from commandline first. This might help you to find the root cause.

tomd135, 2023-06-17:
Hi dehsgr,
sorry for not answering, was without access for a few days.
My TWCmanager is running dedicated on a Raspi 4.
Tried curl on commandline:
`pi@raspberrypi:~ $ curl -k https://192.168.123.102:1881/solar/consumption
-6163.9pi@raspberrypi:~ $ curl -k https://192.168.123.102:1881/solar/generation
7152pi@raspberrypi:~ $  `
Return values are looking good from my point of view.
On the TWCmanager page, generation and consumption are still 0.
Any other thoughts?
Thanks Thomas

dehsgr, 2023-06-17:
@tomd135 you might post your full configuration here. Maybe there is sth. wrong with it. I‘m running that module without any issue.

tomd135, 2023-06-17:
without option -k
[image]

dehsgr, 2023-06-17:
Yeah. That‘s your issue. You would have to ensure trust to that certificate. That might help you: https://unix.stackexchange.com/questions/90450/adding-a-self-signed-certificate-to-the-trusted-list

tomd135, 2023-06-23:
@dehsgr, After several tries I am getting now consumption and generation data.
`Jun 23 17:15:46 raspberrypi python3[8047]: 17:15:46 ⛽ Manager  20 Green energy Generates 4497W, Consumption 1236W (Charger Load 0W, Other Load 1236W)`.
The general issue is of course like you mentioned earlier the self signed nodered certificate for the https connection.
I still have warning in my log:
`Jun 23 17:19:49 raspberrypi python3[8047]: /usr/local/lib/python3.9/dist-packages/urllib3/connection.py:463: SubjectAltNameWarning: Certificate for venus.local has no `subjectAltName`, falling back to check for a `commonName` for now. This feature is being removed by major browsers and deprecated by RFC 2818. (See https://github.com/urllib3/urllib3/issues/497 for details.)
Jun 23 17:19:49 raspberrypi python3[8047]:   warnings.warn(
Jun 23 17:19:49 raspberrypi python3[8047]: /usr/local/lib/python3.9/dist-packages/urllib3/connection.py:463: SubjectAltNameWarning: Certificate for venus.local has no `subjectAltName`, falling back to check for a `commonName` for now. This feature is being removed by major browsers and deprecated by RFC 2818. (See https://github.com/urllib3/urllib3/issues/497 for details.)
Jun 23 17:19:49 raspberrypi python3[8047]:   warnings.warn(` 
Any idea how to solve this or get rid of?
Thanks Thomas

dehsgr, 2023-06-23:
Thanks for getting back. Nice that you got it running. To solve the last issue you would have to ensure that your certificate has the subject alternative name attribute set. The attribute value should reflect the common name of your certificate. See eg. OpenSSL documentation about getting this done.