ngardiner/TWCManager #527: Track Green Energy is overriding the Non-Scheduled power action

alexeiw123, 2023-06-23:
I have Non-Scheduled power action set to 'Do Not Charge' and scheduled charging set from midnight to 6am.
It seems the scheduled setting to resume green tracking takes precedent over the 'do not charge' setting.

With TWCManger starting charge from the API, this means it causes the car to charge despite the car having scheduled charging set and TWC manager set to do not charge.

[image]

[image]

[image]

Note that setting the scheduled resume green energy setting to a time after sunset, or creating a large negative generation offset works around this issue.

MikeBishop, 2023-07-03:
This looks to be the expected behavior, albeit probably poor naming.  Your Scheduled Charging time is 0:00 to 6:00, and Track Green Energy will run from the later of sunrise or the Resume Tracking time until sunset; you have it set to 0:00, so sunrise is always later.

The Non-Scheduled policy is effectively the end of the policy list, when nothing else is active.  Your workarounds are reasonable; you could also simply disable monitoring of generation data if you're not using it to drive the Track Green Energy policy.

alexeiw123, 2023-07-04:
> This looks to be the expected behavior, albeit probably poor naming. Your Scheduled Charging time is 0:00 to 6:00, and Track Green Energy will run from the later of sunrise or the Resume Tracking [...]

Oh OK that makes sense. It's hard to communicate succinctly through the UI what the expected behaviour is. Perhaps a graphical timeline could work , like what Tesla uses for the tariffs within it's charge tracking app features?

[image]

I've landed on a setting that's working for me now anyway. my super-off peak is 8c/kWh and my feed in is 10c/kWh. The cost difference of 2c/kWh is negligible so I'm still favouring green energy for charging up to 90% but will charge the car from super off-peak if SoC is low. Seems to be working ok!

MikeBishop, 2023-07-05:
That's the sort of scenario that adding SoC limits to the charging policies was intended for -- a Scheduled Charging limit of 50% and a TGE limit of 90%, for example.

ngardiner, 2026-04-26:
Fixed in cb501d8