ngardiner/TWCManager #528: VIN not available for 2 of 3 TWC

chiefjoejj, 2023-06-25:
Hi there! Just one quick question:
I have three TWC installed separately but only one device is telling me the VIN of the connected Tesla...

The one showing the VIN has the ID 8906 - the two devices that don't show the VIN have the ID 2013 and 7835 (so older production date I guess).

Perhaps someone can give me a hint wether I am doing something wrong - or is this feature only working with "newer" IDs from 8000 up or something? I use TWCManager v1.2.3 - each TWC connected separately via its own RS485-USB-adapter and docker container with the same settings.

MikeBishop, 2023-07-03:
I don't know the cutoffs of production dates, but I think it has been noted that older TWCv2 units don't report the VIN of the connected car. If Tesla were of a mind to push firmware updates, they probably could, but at this point it's safe to assume they won't do that.

Assuming that the lack-of-VIN stays with the TWC and not the vehicle, that's probably what you're seeing.

chiefjoejj, 2023-07-03:
Hi Mike,
thats true. I think I will give it a try with another TWC charger on eBay.

My "lucky guess" is the following:
Models with the printed "(S)TSN" number starting with "A18F..." show the VIN, "A18E..." don't. Perhaps you or someone else could be so kind to check their TWC for reference...

Best regards
Jay

PS: My TWC that doesn't show the VIN is reacting to the ampere-change-command correctly but doesn't react to the start/stop-command at all. Whereas my VIN-reporting-TWC works like a charm with that.

pirinex, 2023-08-28:
In my case one it's showing VIN and the other one nothing... But the one it's showing VIN,  is #3123 and the #6856 (this one being DC for Tesla only) doesn't show any VIN. And both are (S)TSN: "A17...")