ngardiner/TWCManager #531: Did 1.3 broke the Kostal inverter plug-in?

danielsan1, 2023-08-27:
Hi,
I just installed 1.3 on a new pi as I wasn’t able to upgrade from 1.2.6 (see previous issue).
Now I got another issue when I try to configure my Kostal inverter into the config.json (I copied it 1:1 from my working 1.2.6 installation)

pi@TWCManager:~ $ sudo docker-compose -f docker-compose.yml up
Creating network "pi_default" with the default driver
Creating pi_twcmanager_1 ... done
Attaching to pi_twcmanager_1
twcmanager_1  | 14:32:24 🎮 HTTP     20 Serving at port: 8080
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
twcmanager_1  |     return _run_code(code, main_globals, None,
twcmanager_1  |   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
twcmanager_1  |     exec(code, run_globals)
twcmanager_1  |   File "/usr/src/TWCManager/TWCManager.py", line 34, in <module>
twcmanager_1  |     import TWCManager.TWCManager
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 608, in <module>
twcmanager_1  |     modinstance = modclassref(master)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/Kostal.py", line 85, in __init__
twcmanager_1  |     inv_model = self.__readModbus(768, "String")
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/Kostal.py", line 109, in __readModbus
twcmanager_1  |     if not self.modbus.is_open():
twcmanager_1  | TypeError: 'bool' object is not callable

Can you help me with that? It has to be connected to 1.3 as a) it’s working on 1.2.3 b) if I leave the Kostal config out of the config.json it works (Docker)

Thanks in advance!

daniel

danielsan1, 2023-09-17:
Hey @MikeBishop & @ngardiner - do you have any ideas what this problem causes? Is it really related to the 1.3 version? 
I tried to get in contact with the plug-in's author but with no luck :-/ Would love to upgrade to v 1.3 

Thanks for your help!

MikeBishop, 2023-09-19:
I'm sorry, off-hand I'm not sure what that would be.  It's difficult to debug plug-ins for which you don't have the appropriate hardware.

tjikkun, 2023-09-19:
You can try to replace `/usr/src/TWCManager/lib/TWCManager/EMS/Kostal.py` line 109:
from
` if not self.modbus.is_open():`
to
` if not self.modbus.is_open:`

It might help, but I don't know for sure.

danielsan1, 2023-09-20:
Hi both! Thank you very much for your replies/ideas. I’ll try it as soon as possible - hopefully at the weekend.

danielsan1, 2023-09-24:
Hi @tjikkun - it worked! Thanks a lot again :)

Unfortunately I ran in two other directly to TWCmanager related problems... 

When I want to **enter (and save) my Tesla token** as asked on the front page I get a "ERR_EMPTY_RESPONSE" in the browser and the following in the terminal:
twcmanager_1  | Exception occurred during processing of request from ('192.168.188.23', 50061)
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
twcmanager_1  |     self.finish_request(request, client_address)
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
twcmanager_1  |     self.RequestHandlerClass(request, client_address, self)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
twcmanager_1  |     BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
twcmanager_1  |     self.handle()
twcmanager_1  |   File "/usr/lib/python3.9/http/server.py", line 427, in handle
twcmanager_1  |     self.handle_one_request()
twcmanager_1  |   File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
twcmanager_1  |     method()
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 985, in do_POST
twcmanager_1  |     res = master.getModuleByName("TeslaAPI").saveApiToken(url)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1058, in saveApiToken
twcmanager_1  |     params = json.loads(req.text)
twcmanager_1  |   File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
twcmanager_1  |     return _default_decoder.decode(s)
twcmanager_1  |   File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
twcmanager_1  |     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
twcmanager_1  |   File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
twcmanager_1  |     raise JSONDecodeError("Expecting value", s, err.value) from None
twcmanager_1  | json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Also, when I want to **set my home address** I get a "ERR_EMPTY_RESPONSE" in the browser and the following in the terminal:
twcmanager_1  | Exception occurred during processing of request from ('192.168.188.23', 50096)
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
twcmanager_1  |     self.finish_request(request, client_address)
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
twcmanager_1  |     self.RequestHandlerClass(request, client_address, self)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
twcmanager_1  |     BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
twcmanager_1  |   File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
twcmanager_1  |     self.handle()
twcmanager_1  |   File "/usr/lib/python3.9/http/server.py", line 427, in handle
twcmanager_1  |     self.handle_one_request()
twcmanager_1  |   File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
twcmanager_1  |     method()
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 941, in do_POST
twcmanager_1  |     self.process_home_location()
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 1141, in process_home_location
twcmanager_1  |     del master.settings["homeLat"]
twcmanager_1  | KeyError: 'homeLat'

Did something went wrong with the Docker installation in general? Do you maybe have another easy fix for this? (or @ngardiner 

Thanks for your time (& ideas)!

Daniel

closed 2023-10-01 by danielsan1